ros2/rviz #847: Wayland Support

ciandonovan-akara, 2022-04-01:
Please enable OGRE EGL support with the CMake flag -DOGRE_GLSUPPORT_USE_EGL=1

ciandonovan, 2022-04-04:
```
$ /usr/bin/podman run --rm -it --net=host --env QT_QPA_PLATFORM=wayland --env XDG_RUNTIME_DIR=/run --volume=/run/user/1000/wayland-0:/run/wayland-0 --device=/dev/dri/card0 rviz_test /bin/bash
root@a4ae111c8866:/# rviz2
Using Wayland-EGL
Using the 'xdg-shell' shell integration
Segmentation fault
```
Without EGL support compiled into OGRE, RViz segfaults when trying to use the Wayland platform.
AFAIK starting with OGRE 1.13 EGL support will be compiled in by default, does ROS2 plan to move to OGRE 1.13 anytime soon?

clalancette, 2022-04-06:
> AFAIK starting with OGRE 1.13 EGL support will be compiled in by default, does ROS2 plan to move to OGRE 1.13 anytime soon?

No, not soon.  Our main platform (Ubuntu 22.04) is still on OGRE 1.12, so we generally follow that.

ciandonovan, 2022-04-12:
Is the RViz that you ship as binary packages as part of an official ROS distro built from the Ubuntu OGRE package or from source? The build instructions on github seem to indicate that that building from source without relying on other Ubuntu ROS packages is preferred.
If it is the case that RViz is built from upstream OGRE, is there any possibility that its build flags can be modified, or is it the case that RViz dynamically links with Ubuntu's OGRE libraries?

clalancette, 2022-04-12:
As it stands, we always vendor our own package (build from source).  The goal is to eventually link against Ubuntu's OGRE libraries, but we aren't ready to do that yet.

ciandonovan, 2022-04-12:
In that case would you be open to tweaking how you build OGRE in order to support features like Wayland, or would you rather keep the OGRE libraries built from upstream as similar to the Ubuntu libraries as possible if the goal is to eventually link against them?

clalancette, 2022-04-21:
> In that case would you be open to tweaking how you build OGRE in order to support features like Wayland, or would you rather keep the OGRE libraries built from upstream as similar to the Ubuntu libraries as possible if the goal is to eventually link against them?

We can tweak it (we already do in other ways), but that tweak might go away if we ever use the upstream package.

victorcarvesk, 2023-04-11:
I'm unable to run RViz2 on Ubuntu 22.04 (using Wayland), what I can do to resolve this?

mosfet80, 2024-07-09:
you need to migrate to the latest version of ogre to fix the problem.
This involves changing rvitz_rendering
I'm working on it