ros2/rviz #852: [rolling] Laserscan status seems to jump

audrow, 2022-04-18:
Platform: Ubuntu Jammy in a Docker container (on an Ubuntu Focal machine)
DDS: FastDDS

The laserscan status seems to jump in and out of the error state, creating a sort of error flashing.

You can find steps to reproduce this here: https://github.com/osrf/ros2_test_cases/issues/412#issue-1200261055

And a comment describing the issue: https://github.com/osrf/ros2_test_cases/issues/412#issuecomment-1095520785

jacobperron, 2022-05-12:
I can reproduce the issue on building from source on Ubuntu 20.04:

```
# terminal 1
ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
# terminal 2
rviz2
```

Change fixed frame to `world` and view `/scan` topic.

clalancette, 2022-05-13:
> I can reproduce the issue on building from source on Ubuntu 20.04:

Oh, that's interesting.  It seemed to go away for me in later Rviz2 packages on Jammy.  Still not sure what is going on there.

jacobperron, 2022-05-13:
I took at look at recent changes in the code path of the laser scan display and found the issue goes away if I revert https://github.com/ros-perception/laser_geometry/pull/83.

jacobperron, 2022-05-13:
I think what's happening is,

1. `laser_geometry` is accounting for the `time_increment` field in the laser scan message by asking for a transform at a time later than the header stamp. I think this assumes that the time stamp of the laser scan message is less than the time when the message was actually sent.
2. The dummy laser sensor [set's a non-zero `time-increment`](https://github.com/ros2/demos/blob/b64830e041dcd6bcd934cf222b53571f84706f6f/dummy_robot/dummy_sensors/src/dummy_laser.cpp#L61-L62), but doesn't delay by the time increment before sending the message, so we get an "extrapolation into the future" lookup error in RViz.

I don't know if the `laser_geometry` assumption about the timestamp and time increment is incorrect, or if we should add a delay to the dummy laser sensor to match the assumption.

jacobperron, 2022-05-13:
Ah, it seems that we do document that the header timestamp is for the _first ray_ in the scan:

https://github.com/ros2/common_interfaces/blob/b7e2ad503e1648468b1dc766c43aac12cc9aee5a/sensor_msgs/msg/LaserScan.msg#L7-L8

So, I think the correct thing to do is fix the dummy sensor.

clalancette, 2023-04-26:
I just tested this on Iron, and as far as I can tell this is fixed.  The laser scan doesn't flicker, and we don't get any repeated output on the console.  So I'm going to close this out, but feel free to reopen if this is still a problem.

closed 2023-04-26 by clalancette