ros2/rviz #856: LNK2019 error using VisualizationManager class on Windows

jonah-gourlay44, 2022-04-28:
I have a class "QtDemo" that has a rviz_common::VisualizationManager* as a private member "manager_". I have a function QtDemo::InitRviz() that initializes the rviz members of the QtDemo class. 
[image]
When I try to compile this function against either the pre-compiled foxy binaries, or an rviz2 source build, I get the following linker errors:
[image]
I am linking against

- rviz_common::rviz_common
- rviz_rendering::rviz_rendering
- rviz_ogre_vendor::OgreMain
- rviz_orgre_vendor::OgreOverlay
- Qt5::Core
- Qt5::Widgets

My full [CMakeLists.txt](https://github.com/ros2/rviz/files/8584850/CMakeLists.txt) for reference. For the source build I verified that visualization_manager.cpp was being compiled and that it had the same function definitions. It seems like this is the only file affected, but I can't figure out why. Has anyone seen this before, or have any idea what could be causing it? I am using ROS2 foxy on Windows 10, and doing the source build from the "foxy" development branch.

clalancette, 2022-04-28:
My best guess here is that VisualizationManager is missing `RVIZ_COMMON_PUBLIC` [here](https://github.com/ros2/rviz/blob/1211b1737a3a4b7f6cd9d148ce073a45059d07bf/rviz_common/include/rviz_common/visualization_manager.hpp#L90).  It would be worthwhile to try adding it to the class and seeing if that makes a difference.  If you can't get that to work, you may instead have to add `RVIZ_COMMON_PUBLIC` to every method inside of the class.

jonah-gourlay44, 2022-04-28:
> My best guess here is that VisualizationManager is missing `RVIZ_COMMON_PUBLIC` [...]

That was it, thank you. The VisualizationManager class just needed the RVIZ_COMMON_PUBLIC macro before it.

closed 2022-04-28 by jonah-gourlay44

clalancette, 2022-05-02:
> That was it, thank you. The VisualizationManager class just needed the RVIZ_COMMON_PUBLIC macro before it.

If you'd like to open a pull request to add that for future users, we'd be happy to review it.