ros2/rviz #857: [embedded Rviz]: change background color

facontidavide, 2022-05-05:
Hi,

I am using Rvix embedded (what was formely used as "librviz") but despite searching in the code, I can't find a way to change the Global Options "Background Color". 

Any suggestion?

jacobperron, 2022-05-19:
To clarify, do you want to change the color via the library API? There's a background property that we can call `setColor` on, however it is a `protected` member:

https://github.com/ros2/rviz/blob/7b4ba855b5803ef8b77b7732889c8b1b6e43532b/rviz_common/include/rviz_common/visualization_manager.hpp#L376

How was this done in ROS 1? The code looks similar:

https://github.com/ros-visualization/rviz/blob/1dc4953639dd69b4a52db746d7101638fa13e6c9/src/rviz/visualization_manager.h#L433

facontidavide, 2022-05-20:
yes, I would like to do it from the API. Not being public, I don't know how to access it.

jacobperron, 2022-05-26:
@facontidavide is it possible to change the color in ROS 1? I'm trying to understand if this is something that's missing when transition to ROS 2 or if it's strictly a new feature request.

Either way, I think we could add a public method for setting the color to `VisualizationManager`; which just updates `background_color_property_`. If you think this accomplishes what you want, I'm happy to review a pull request :)

facontidavide, 2022-05-30:
Probably not possible in ROS1 either.

jacobperron, 2022-05-31:
@facontidavide Would you like to propose a pull request?