ros2/rviz #870: RViz

henrygerardmoore, 2022-06-21:
### Description

When loading a config file (.rviz) with RViz, the loader does not respect links set to not display visually. The file is saved properly but on load all the links are visible again.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Binary build

### Steps to reproduce
Load a robot in RViz, toggle visibility of some of the links. Then save, re-load config file, and the links will be visible again despite saving with links correctly set to value false.
[Example configuration file](https://gist.github.com/henrygerardmoore/82dc7200a0ad99b1b59bd22fa2aca46d) with many links toggled to be invisible.

If you then save the file immediately after reloading, the output is different from the input. 

This was set up by using the panda arm from [moveit_resources](https://github.com/ros-planning/moveit_resources) with `ros2 launch moveit_resources_panda_moveit_config demo.launch.py`

### Expected behaviour
When reloading a configuration, all links with value false in the rviz config under the Scene Robot section will not be visible.

### Actual behaviour
All links are visible on loading, whether or not they were set to be visible.

jacobperron, 2022-08-30:
It's not clear from the issue description, but I think the issue is with respect to the moveit_rviz_plugin/PlanningScene plugin?

I tried playing around with the links in the default TF plugin, and those seem to load their state correctly from an RViz config file. I suspect the issue is in the moveit plugin (which is not part of this repository). I don't know where the code is, otherwise I would check, but I would check that it is parsing the config in the Load method, e.g.

https://github.com/ros2/rviz/blob/4e976dc444e7e59ec9cdef30f1f04e0beafcafc8/rviz_default_plugins/src/rviz_default_plugins/displays/tf/tf_display.cpp#L181-L190

---

I'm going to close this ticket since I don't believe it is an issue with RViz itself. If you believe I'm mistaken, we can re-open the ticket.

closed 2022-08-30 by jacobperron

jacobperron, 2022-08-30:
I think I found the relevant section of code,

https://github.com/ros-planning/moveit2/blob/c134d89710e831ae58a66e53d87a78b44b0b307c/moveit_ros/visualization/planning_scene_rviz_plugin/src/planning_scene_display.cpp#L677-L680

I think the plugin should be responsible for checking the `config` object and restore it's state accordingly.

mohitydv09, 2026-03-10:
I am experiencing the same issue described here, but **without using MoveIt or the PlanningScene plugin**. The problem occurs with the standard `rviz_default_plugins/RobotModel` display.

#### Setup

* ROS 2: Humble
* OS: Ubuntu 22.04
* RViz display: `rviz_default_plugins/RobotModel`
* Robot description source: topic (`/robot_description` and `/right_arm/robot_description`)
* No MoveIt / PlanningScene involved

#### Description of the problem

RViz correctly **saves per-link visualization settings** in the `.rviz` config file, but those settings are **not restored when the config is loaded**.

Examples of settings that are saved but not respected after reload:

* `Alpha` for individual links
* `Show Axes` for specific links
* `Value` (link visibility toggle)
* UI state such as expanded/collapsed link tree

The saved `.rviz` file clearly contains the expected values, e.g.:

```
Links:
  link_6:
    Alpha: 0
    Show Axes: true
    Show Trail: false
    Value: true
```

However, when RViz is restarted with the same config:

```
rviz2 -d my_config.rviz
```

the link appears with default settings instead of the saved ones.

For example:

* `link_6` becomes visible again even though `Alpha: 0` was saved.
* `Show Axes` for a link is not enabled even though it was saved as `true`.
* The `Links` tree that was expanded in the saved config appears collapsed after reload.

If I manually toggle the property again in the RViz GUI (e.g., enable `Show Axes` or adjust `Alpha`), it immediately works as expected.

#### Observed behavior

The `.rviz` config file appears to be **saved correctly**, but the `RobotModel` display does not consistently restore those values after loading.

This seems to occur when the robot model is reconstructed from the incoming robot description topic.

#### Expected behavior

When loading an `.rviz` configuration file, the `RobotModel` display should restore the saved per-link properties exactly as written in the config.

#### Minimal reproduction

1. Start RViz with a robot model loaded from `/robot_description`.
2. Expand the `RobotModel → Links` tree.
3. Change properties for a specific link:

   * set `Alpha = 0`
   * enable `Show Axes`
4. Save the configuration.
5. Restart RViz with the saved config.

Expected: link remains invisible and axes remain enabled.
Actual: link appears visible again and axes are disabled.

If needed I can provide a minimal `.rviz` file demonstrating the issue.