ros2/rviz #872: Link inertia error should tell which link has inertia error

SammyRamone, 2022-06-27:
When displaying a robot model one might get the following error
`The link is static or has unrealistic inertia, so the equivalent inertia box will not be shown.`
which is coming from [here](https://github.com/ros2/rviz/blob/7b4ba855b5803ef8b77b7732889c8b1b6e43532b/rviz_default_plugins/src/rviz_default_plugins/robot/robot_link.cpp#L888-L890) 
Unfortunately, the name of the corresponding link is not given, so it is difficult to see which of the links is meant. For complex robot models, it is quiet cumbersome to search for this manually.
So I would propose to provide a message like this
`The link base_link is static or has unrealistic inertia, so the equivalent inertia box will not be shown.`

closed 2022-08-31 by clalancette

andrestoga, 2022-11-29:
Is there a way to know which link is referring to? I installed ROS humble using DEB binaries and I'm not sure how to add the fix for this.

SammyRamone, 2022-12-02:
If I see it correctly, the fix (https://github.com/ros2/rviz/pull/874) was only merged into rolling. Therefore, it is not in Humble. You can either use ROS Rolling or build Humble from scratch and apply the merged PR locally.

cjstahoviak, 2024-08-11:
Can this be pushed into Humble as well?

PR #874 (2022-08-31): Show link names in inertia error message
Closes #872.
PR #1259 (2024-08-19): Show link names in inertia error message (backport #874)
Closes #872.<hr>This is an automatic backport of pull request #874 done by [Mergify](https://mergify.com).