ros2/rviz #884: Size Transformer for PointCloud2 Display

dbruggner, 2022-08-16:
Add a size transformer for pointcloud2 display plugin to make the size of points dependent on specific field in the pointcloud.
As a user of rviz I want to visualize different point properties. Currently there is the possibility to use the color transformer for this. But it's only possible to visualize one single property of the pointcloud. If you also could use the size to visualize a second property it would be very helpful in some scenarios. E.g. visualizing the z coordinate with different colors and the intensity as the size of the individual points.

clalancette, 2022-08-16:
That seems like a reasonable feature request to me.  It's unlikely we are going to have time to work on it anytime soon, but if you find the time do feel free to open a PR; we are happy to review.

nicolaloi, 2024-09-25:
@clalancette is this issue still relevant? I would like to try to implement this new feature and I had a look at the current code. I was thinking something like passing a custom `size` attribute for each vertex to the vertex shader [point.vert](https://github.com/ros2/rviz/blob/14.2.6/rviz_rendering/ogre_media/materials/glsl120/point.vert) instead of using a common `size`, but this might have performance implications, which could make the change not worth it.