ros2/rviz #885: Render System Missing Destructor - Foxy Branch

bchurch8, 2022-08-16:
The Render System class has no destructor on the Foxy branch, which causes various issues when trying to shutdown and then restart rviz in the same process. Additionally, the Render System allocates memory for the Ogre Root and Overlay System, but never deallocates the memory. I noticed that a destructor was implemented on the Rolling branch. Will there be a similar patch for Foxy at any point in the future? 

Specifically, the issue I ran into occurred when using a Rviz Render Panel and Visualization Manager. In our app an rviz display window is created, used for some time, closed and then can be created again. Everything works as expected, until we tried to generate the display window the second time. During the second Visualization Manager constructor, Ogre threw an exception because of a duplicated object name. 

Since the Ogre Root is a singleton it has to be explicitly destroyed, which isn't happening anywhere to my knowledge. As such, when the Visualization manager was constructed and tried to initialize the Rviz colours in Ogre, Ogre threw an exception because these colours already existed. 

Since the Render System created the Ogre Root (and Overlay System), it should also destroy them on shutdown (and the Render System itself should be able to be destroyed). I'm not sure where/if the Render System should be destroyed. In our case the Render System was initialized from the RenderWindowImpl, but I'm not sure if destroying it in the RenderWindowImpl is the proper solution as I don't know where else it may be used.

clalancette, 2022-09-01:
> The Render System class has no destructor on the Foxy branch, which causes various issues when trying to shutdown and then restart rviz in the same process. Additionally, the Render System allocates [...]

We'd definitely be happy to review a PR that adds in a destructor for Foxy.  Keep in mind that the change needs to be backwards-compatible in both API and ABI, so as long as that is the case we can consider it for inclusion.

clalancette, 2023-07-14:
Given that Foxy is now End-of-Life, and this issue is fixed in the latest, I'm going to close this out.  Feel free to open a new issue if you are still running into problems.

closed 2023-07-14 by clalancette