ros2/rviz #886: Release rviz_assimp_vendor as independent assimp_vendor package?

ijnek, 2022-08-21:
The ``rviz_assimp_vendor`` package is used to build assimp on platforms that don't have a system dependency available for assimp.

According to [ROS Index](https://index.ros.org/d/assimp/), assimp is used across at least 5 released packages in rolling, so I believe there is benefit in moving this out to a standalone assimp_vendor package.

Are there some reasons against this move, apart from having greater control over assimp by keeping it rviz specific?

clalancette, 2022-10-28:
I'd be for moving this to general vendored packages.  While we are at it, I'd love to make them use the system versions when they can.

ijnek, 2022-10-30:
Great, what's the best way to proceed? I've gone ahead and moved the package out into its own repo already - https://github.com/ijnek/assimp_vendor. Should we aim to transfer this repo to ros2, or ..?