ros2/rviz #887: Error displaying bayer_rggb8 encoded images.

rshanor, 2022-08-23:
I am using this package to stream images off a basler camera: https://github.com/basler/pylon-ros-camera/tree/galactic

The above package is outputting images in the format `encoding: bayer_rggb8`

When I try to view these images in rviz2, there is a decoding bug. 

[image]

I am able to confirm the issue is in rviz, as the images appear correct under `ros2 run rqt_image_view rqt_image_view`

[image]

Related: https://github.com/basler/pylon-ros-camera/issues/84

duasob, 2024-11-15:
Hi! I seem to have the same problem - where you able to solve this?

lukasj-imar, 2025-04-05:
any update to this problem?

JasperTan97, 2025-11-26:
https://wiki.ros.org/rviz/DisplayTypes/Image

It's just not supported.