ros2/rviz #916: Ros2 Humble/Rolling - rviz2 wont accept my tf_static message from bag

geoeo, 2022-10-15:
This is new in humble/rolling, since I dont have this issue on galactic.

When recording /tf_static via rosbag interface in python (Default QoS). Rviz2 wont accept my tf_static messages when playing the bag bag. It works at runtime via StaticTransfromBroadcaster.

The warning I get is 

```
I get [WARN] [1665740292.480244928] [rosbag2_player]: New subscription discovered on topic '/tf_static', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: DURABILITY_QOS_POLICY
```

Did the expected QoS change/ is more strictly enforced in humble/rolling?

joeldushouyu, 2023-07-17:
I also have a similar issue. Is there any know issue related to this?

haider8645, 2023-07-25:
Hello Guys, I got it to to work.

1. Create a yaml file with the following content:

[image]

2. When playing the bag file, you can override the topic QOS:

`ros2 bag play kitti_2011_09_26_drive_0014_synced  --qos-profile-overrides-path reliability_override.yaml`

Further info:

1. http://docs.ros.org.ros.informatik.uni-freiburg.de/en/crystal/Tutorials/Ros2bag/Overriding-QoS-Policies-For-Recording-And-Playback.html

1. https://design.ros2.org/articles/qos.html

roboav8r, 2024-08-11:
Alternatively, if you're playing back the bag using a [reader object like the tutorials here](https://mcap.dev/guides/python/ros2), then you can create the QoS object and add it as an argument for the `tf_static` publisher:

```
tf_static_qos = rclpy.qos.QoSProfile(history = rclpy.qos.HistoryPolicy.KEEP_LAST, 
                                     depth = 10,
                                     reliability = rclpy.qos.ReliabilityPolicy.RELIABLE,
                                     durability = rclpy.qos.DurabilityPolicy.TRANSIENT_LOCAL)
tf_static_pub = node.create_publisher(TFMessage, 'tf_static', tf_static_qos)
```