ros2/rviz #918: Image Display is auto-disabled after minimazing main window

kudoukidd, 2022-11-01:
Image Display is auto-disabled after minimazing main window in ROS2 Galactic/Humble, which is different from ROS1 Melodic/Noetic

clalancette, 2022-11-17:
Can you please be more specific?  What's the behavior that you are seeing, and what do you expect to happen?  Screenshots might help here.

kudoukidd, 2022-11-17:
void Display::setAssociatedWidget(QWidget * widget)
{
  if (associated_widget_panel_) {
    disconnect(
      associated_widget_panel_, SIGNAL(visibilityChanged(bool)), this,
      SLOT(associatedPanelVisibilityChange(bool)));
    disconnect(associated_widget_panel_, SIGNAL(closed()), this, SLOT(disable()));
  }

  associated_widget_ = widget;
  if (associated_widget_) {
    WindowManagerInterface * wm = context_->getWindowManager();
    if (wm) {
      associated_widget_panel_ = wm->addPane(getName(), associated_widget_);
      **connect(
        associated_widget_panel_, SIGNAL(visibilityChanged(bool)), this,
        SLOT(associatedPanelVisibilityChange(bool)));**
      connect(associated_widget_panel_, SIGNAL(closed()), this, SLOT(disable()));
      associated_widget_panel_->setIcon(getIcon());
    } else {
      associated_widget_panel_ = nullptr;
      associated_widget_->setWindowTitle(getName());
    }
  } else {
    associated_widget_panel_ = nullptr;
  }
}

There is a connection between associated_widget_panel_ RenderPanel visibility and Display "enability". When rviz main window is minimized by clicking minimize button, which makes ImageDisplay RenderPanel unvisible, the ImageDisplay would become disabled and has to be manually re-enabled by checking the ImageDisplay enable property.

There is a suppress_hiding_associated_widget_panel_ flag in rviz1 which is different from rviz2. I did not dig further because simply removing the connection works fine in my situation.

MarcoMagriDev, 2023-01-18:
Same issue here. The same problem occurs also with the Camera Display plugin.