ros2/rviz #921: Dynamic linking error when installed through apt, unless setup file sourced

ijnek, 2022-11-04:
There is a dynamic linking error to libOgre when rviz2 is installed through apt. The workaround is to source the distro's setup.bash file. Preliminary question, but does one have to run ``source /opt/ros/${ROS_DISTRO}/setup.bash`` manually after installing a binary ros package?

To reproduce problem:
```sh
# Run humble docker container
docker run -it ros:humble

# Inside docker container, install rviz2
apt update && apt install ros-humble-rviz2

# Run rviz2
rviz2
```

Output:

```sh
rviz2: error while loading shared libraries: libOgreMain.so.1.12.1: cannot open shared object file: No such file or directory
```

Running ``ldd``:

```
root@abd1c1c02988:/# ldd /opt/ros/humble/lib/rviz2/rviz2 
	linux-vdso.so.1 (0x00007ffda77e8000)
	librviz_common.so => /opt/ros/humble/lib/librviz_common.so (0x00007fd3046e4000)
	libQt5Widgets.so.5 => /lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007fd30401c000)
	libQt5Core.so.5 => /lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007fd303abd000)
	librclcpp.so => /opt/ros/humble/lib/librclcpp.so (0x00007fd3038e5000)
	librcutils.so => /opt/ros/humble/lib/librcutils.so (0x00007fd3038cd000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fd30369f000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fd30367f000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fd303457000)
	libOgreMain.so.1.12.1 => not found
	librviz_rendering.so => /opt/ros/humble/lib/librviz_rendering.so (0x00007fd3033ff000)
	libyaml-cpp.so.0.7 => /lib/x86_64-linux-gnu/libyaml-cpp.so.0.7 (0x00007fd3033c0000)
	libQt5Gui.so.5 => /lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007fd302ce0000)
	libresource_retriever.so => /opt/ros/humble/lib/libresource_retriever.so (0x00007fd302cd9000)
	librcl.so => /opt/ros/humble/lib/librcl.so (0x00007fd302c9b000)
	libament_index_cpp.so => /opt/ros/humble/lib/libament_index_cpp.so (0x00007fd302c90000)
	libclass_loader.so => /opt/ros/humble/lib/libclass_loader.so (0x00007fd302c7d000)
	librcpputils.so => /opt/ros/humble/lib/librcpputils.so (0x00007fd302c6f000)
	libtinyxml2.so.9 => /lib/x86_64-linux-gnu/libtinyxml2.so.9 (0x00007fd302c55000)
	libconsole_bridge.so.1.0 => /lib/x86_64-linux-gnu/libconsole_bridge.so.1.0 (0x00007fd302c4f000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fd302b68000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fd30487c000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fd302b4c000)
	libdouble-conversion.so.3 => /lib/x86_64-linux-gnu/libdouble-conversion.so.3 (0x00007fd302b37000)
	libicui18n.so.70 => /lib/x86_64-linux-gnu/libicui18n.so.70 (0x00007fd302806000)
	libicuuc.so.70 => /lib/x86_64-linux-gnu/libicuuc.so.70 (0x00007fd30260b000)
	libpcre2-16.so.0 => /lib/x86_64-linux-gnu/libpcre2-16.so.0 (0x00007fd302581000)
	libzstd.so.1 => /opt/ros/humble/lib/libzstd.so.1 (0x00007fd3024f2000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007fd3023b8000)
	liblibstatistics_collector.so => /opt/ros/humble/lib/liblibstatistics_collector.so (0x00007fd3023b1000)
	librcl_interfaces__rosidl_typesupport_cpp.so => /opt/ros/humble/lib/librcl_interfaces__rosidl_typesupport_cpp.so (0x00007fd3023a4000)
	librcl_yaml_param_parser.so => /opt/ros/humble/lib/librcl_yaml_param_parser.so (0x00007fd302398000)
	librosgraph_msgs__rosidl_typesupport_cpp.so => /opt/ros/humble/lib/librosgraph_msgs__rosidl_typesupport_cpp.so (0x00007fd302393000)
	libstatistics_msgs__rosidl_typesupport_cpp.so => /opt/ros/humble/lib/libstatistics_msgs__rosidl_typesupport_cpp.so (0x00007fd30238e000)
	libtracetools.so => /opt/ros/humble/lib/libtracetools.so (0x00007fd302389000)
	librmw_implementation.so => /opt/ros/humble/lib/librmw_implementation.so (0x00007fd30237c000)
	librcl_logging_interface.so => /opt/ros/humble/lib/librcl_logging_interface.so (0x00007fd302375000)
	librmw.so => /opt/ros/humble/lib/librmw.so (0x00007fd302369000)
	libOgreMain.so.1.12.1 => not found
	libOgreOverlay.so.1.12.1 => not found
	libassimp.so.5 => /lib/x86_64-linux-gnu/libassimp.so.5 (0x00007fd301c23000)
	libGLX.so.0 => /lib/x86_64-linux-gnu/libGLX.so.0 (0x00007fd301bed000)
	libX11.so.6 => /lib/x86_64-linux-gnu/libX11.so.6 (0x00007fd301aad000)
	libGL.so.1 => /lib/x86_64-linux-gnu/libGL.so.1 (0x00007fd301a26000)
	libpng16.so.16 => /lib/x86_64-linux-gnu/libpng16.so.16 (0x00007fd3019eb000)
	libharfbuzz.so.0 => /lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007fd30191c000)
[... 134 lines cut ...]
	libspdlog.so.1 => /lib/x86_64-linux-gnu/libspdlog.so.1 (0x00007f3389a03000)
	librosidl_typesupport_c.so => /opt/ros/humble/lib/librosidl_typesupport_c.so (0x00007f338a40c000)
	libbuiltin_interfaces__rosidl_generator_c.so => /opt/ros/humble/lib/libbuiltin_interfaces__rosidl_generator_c.so (0x00007f33899fe000)
	libXau.so.6 => /lib/x86_64-linux-gnu/libXau.so.6 (0x00007f33899f8000)
	libXdmcp.so.6 => /lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f33899f0000)
	libunistring.so.2 => /lib/x86_64-linux-gnu/libunistring.so.2 (0x00007f3389846000)
	libgnutls.so.30 => /lib/x86_64-linux-gnu/libgnutls.so.30 (0x00007f3389659000)
	libhogweed.so.6 => /lib/x86_64-linux-gnu/libhogweed.so.6 (0x00007f3389611000)
	libnettle.so.8 => /lib/x86_64-linux-gnu/libnettle.so.8 (0x00007f33895cb000)
	libgmp.so.10 => /lib/x86_64-linux-gnu/libgmp.so.10 (0x00007f3389549000)
	libkrb5.so.3 => /lib/x86_64-linux-gnu/libkrb5.so.3 (0x00007f338947e000)
	libk5crypto.so.3 => /lib/x86_64-linux-gnu/libk5crypto.so.3 (0x00007f338944d000)
	libcom_err.so.2 => /lib/x86_64-linux-gnu/libcom_err.so.2 (0x00007f3389447000)
	libkrb5support.so.0 => /lib/x86_64-linux-gnu/libkrb5support.so.0 (0x00007f3389439000)
	libsasl2.so.2 => /lib/x86_64-linux-gnu/libsasl2.so.2 (0x00007f338941e000)
	libbrotlicommon.so.1 => /lib/x86_64-linux-gnu/libbrotlicommon.so.1 (0x00007f33893fb000)
	libfmt.so.8 => /lib/x86_64-linux-gnu/libfmt.so.8 (0x00007f33893da000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f33893c0000)
	libp11-kit.so.0 => /lib/x86_64-linux-gnu/libp11-kit.so.0 (0x00007f3389285000)
	libtasn1.so.6 => /lib/x86_64-linux-gnu/libtasn1.so.6 (0x00007f338926d000)
	libkeyutils.so.1 => /lib/x86_64-linux-gnu/libkeyutils.so.1 (0x00007f3389266000)
	libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x00007f3389252000)
	libmd.so.0 => /lib/x86_64-linux-gnu/libmd.so.0 (0x00007f3389243000)
	libffi.so.8 => /lib/x86_64-linux-gnu/libffi.so.8 (0x00007f3389236000)
```

clalancette, 2022-11-04:
> Preliminary question, but does one have to run `source /opt/ros/${ROS_DISTRO}/setup.bash` manually after installing a binary ros package?

In the general case, yes.  You need to source it for exactly this reason; you don't know exactly where the libraries may be installed.  You'll only find that out by sourcing the setup.bash.

The caveat is that in many cases, it works without sourcing since a large portion of our libraries *are* installed to `/opt/ros/humble/lib`, and that path was likely added by something else.  But it is not guaranteed, and thus you always need to source after installing.

ijnek, 2022-11-04:
Thanks @clalancette, your answer was very helpful and I hope others also find IT too! Closing as this isn't a bug.

closed 2022-11-04 by ijnek