ros2/rviz #925: rviz_ogre_vendor malformed patch error

jollypolly123, 2022-11-22:
Not sure if I'm asking in the right place, but hoping I could get some guidance on how to resolve this.

I'm following this guide to setup ROS 2 on MacOS 13.0: https://docs.ros.org/en/humble/Installation/Alternatives/macOS-Development-Setup.html

Running `colcon build --symlink-install --packages-skip-by-dep python_qt_binding` gives me this error so the build fails:
```
patch: **** malformed patch at line 1817: 
make[2]: *** [ogre-v1.12.1-prefix/src/ogre-v1.12.1-stamp/ogre-v1.12.1-patch] Error 2
make[1]: *** [CMakeFiles/ogre-v1.12.1.dir/all] Error 2
make: *** [all] Error 2
```

closed 2022-11-28 by clalancette

jacetto, 2022-11-29:
I still get the same error, even after removing the build and install dirs

openvmp, 2023-01-06:
Same here. The problem remains even after removing the build and install dirs.
It happens both with the latest and an old (~6m old) Humble source trees.
It started happening after MacOS (and XCode) were updated.

openvmp, 2023-01-06:
@clalancette , PTAL

Kaju-Bubanja, 2023-11-07:
Will this be backported to humble?