ros2/rviz #929: Impossible to run Rviz2 from a Docker container on Apple Silicon

ygoumaz, 2022-11-30:
The Webots team is currently implementing a Docker solution for users on macOS so that they can use ROS 2 packages more easily and in particular our `webots_ros2` package. However we are experiencing some issues with Rviz2. On our macOS M1 machine, it seems impossible to launch Rviz2 inside a `arm64v8/ros:humble-perception` Docker container and show the GUI on the host display.

Here is how to reproduce the issue:

1. Install Docker on the macOS device (https://docs.docker.com/desktop/install/mac-install/).
2. Install XQuartz with `brew install xquartz` in a terminal.
3. Open XQuartz with `open -a xquartz`.
4. In the menu bar, go to `XQuartz > Preferences... > Security > Allow connection from Network clients`
5. Add Path to "xhost": `export PATH=/usr/X11/bin/xhost:$PATH` in the terminal.
6. Enable connections with xhost: `xhost +` in the terminal.
7. Launch humble Docker container with: `docker run -d --name rviz_issue -e DISPLAY=host.docker.internal:0 -v /tmp/.X11-unix:/tmp/.X11-unix -it arm64v8/ros:humble-perception`.
8. Run `docker ps` to get the ID of the launched container.
9. Enter the container with `docker exec -it <container_ID> bash`.
10. Inside the container run `sudo apt update && sudo apt install ros-humble-desktop`.
11. Source ROS 2: `source /opt/ros/humble/setup.bash`.
12. run rviz2 with `rviz2`.

Rviz seems to launch but a series of messages indicate that the OpenGL rendering failed with the following main error message: `RenderingAPIException: OpenGL 1.5 is not supported in GLRenderSystem::initialiseContext`.

I tried to launch rviz1 and other GUI applications, like synaptic and it works fine. So it seems to be an issue related to Rviz. Is this an known issue? Do you have any idea if there is a way to fix this?

clalancette, 2022-11-30:
> Is this an known issue?

I don't think we know about it, no.  Apple Silicon isn't one of our supported platforms, so we don't test it there.

> Do you have any idea if there is a way to fix this?

I'm sure there is a way, but somebody needs to spend the time to investigate.  If I was going to investigate it, I would start by looking at OGRE that we are vendoring (1.12.10).  It's been a while since I looked at it, but I believe there were some assumptions in there about APPLE and !ARM64 which may not be valid anymore.

ygoumaz, 2022-12-01:
Thanks for your answer. 

> Apple Silicon isn't one of our supported platforms, so we don't test it there.

I tested on an Intel chip macOS machine and can reproduce the exact same problem. So it doesn't seem to come from the architecture.

>  If I was going to investigate it, I would start by looking at OGRE that we are vendoring (1.12.10).

Error messages mention OGRE, so I will investigate in this direction.

ygoumaz, 2022-12-07:
I have investigated the issue a little more and found out another open issue on the XQuartz repository which seems to be highly similar to this one: https://github.com/XQuartz/XQuartz/issues/144.

Apparently it seems to be a problem with XQuartz or Dockers in general on macOS. The OpenGL version available in the Docker is only 1.4, while higher versions are required by OGRE or other softwares. In fact `glxgears` and other apps that I was able to run do not require high versions of OpenGL. 

So the problem probably doesn't come from RViz and would be solved once higher OpenGL versions could be forwarded to containers.

danem, 2023-01-18:
@ygoumaz is this not a limitation of GLX itself? I believe the protocol only supports [Opengl up to 1.4](https://www.scm.com/doc/Installation/Remote_GUI.html#opengl-direct-or-indirect-rendering)

bwingo47, 2023-02-11:
@ygoumaz ,
> I tried to launch rviz1 and other GUI applications, like synaptic and it works fine.

How did you get rviz1 to work? I have the same setup as you (docker container on M1 Mac running Ventura 13.2). Rviz1 and Gazebo always give the following error: 
> libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast

Tried everything from #ros-visualization/rviz/issues/1672 and #XQuartz/XQuartz/issues/144, but nothing works. This seems to be a XQuartz issue. Are there any alternative ways to forward GUI from docker containers without XQuartz?

z-learner, 2023-03-13:
I am looking forward answer too....

ygoumaz, 2023-03-13:
@bwingo47 I was able to fix these messages with by setting LIBGL_ALWAYS_INDIRECT=1. The following blog post also explains how to configure XQuartz to enable indirect GLX: http://notepad2.blogspot.com/2018/07/enable-iglx-on-mac-os-xquartz-or-linux.html.

bwingo47, 2023-03-14:
@ygoumaz thank you for the reply. I agree with your conclusions here. Instead of forcing XQuartz to work properly on Apple Silicon, I ended up pursuing the software rendering route, using X11 virtual framebuffer and noVNC web viewer. I hacked together a working docker container ([here](https://github.com/bwingo47/docker_envs/tree/mac_arm_dev/mac_env)) for Apple Silicon Macs by following this [noVNC display container](https://github.com/theasp/docker-novnc). 

The performance is quite good when running Rviz 1 on my 13" M1 MBP (no noticeable stutter or delay), and I'm pretty sure this setup will work with Rviz 2 as well (although not tested). Feel free to open any issues on my repo, I know it's quite cluttered right now, and maybe a little hard to use. 

@z-learner maybe take a look at my [work-around](https://github.com/bwingo47/docker_envs/tree/mac_arm_dev/mac_env) for Apple Silicon Macs, which bypasses XQuartz entirely.

chipsnyder, 2023-03-14:
@bwingo47 Thank you for posting this update. I was able to use the [noVNC display container](https://github.com/theasp/docker-novnc) to get rviz2 to display on my M1.

closed 2023-04-04 by ygoumaz

D4ve0s, 2023-08-15:
I just experienced this issue running  Docker [ros:humble-ros-base-jammy] on Windows 11 using VsCode Remote Dev Container and VcXsrv installed on Windows side.  
RQT_GRAPH works fine. 
RVIZ2 gives `[rviz2]: RenderingAPIException: OpenGL 1.5 is not supported in GLRenderSystem::initialiseContext`

Getting ROS2 "in any variant" working on Windows or Raspberry PI has been a major hassle. It's taken a week just to get working Docker containers on both platforms....issue after issue after issue....I'm so close to actually writing my first node,  a week later :-)

Anyway, what worked for me was to change LIBGL_ALWAYS_INDIRECT=1 to LIBGL_ALWAYS_SOFTWARE=1 which appears to use the latest OpenGL 4.5 version.  

From what I can tell (I am no expert) LIBGL_ALWAYS_INDIRECT causes the use of the native OpenGL library [WGL (Wiggle) ](https://en.wikipedia.org/wiki/WGL_(API)) to interface OpenGL to Windows UI. Clearly this presents an outdated interface/OpenGL version.  Perhaps MacOS's equivalent CGL does a similar back versioning? 

devcontainer.json:
```
"containerEnv": {
    "DISPLAY": "host.docker.internal:0",
    "XDG_RUNTIME_DIR": "/tmp/runtime-${USER}",
    // "LIBGL_ALWAYS_INDIRECT": "1"   // Bad juju
    "LIBGL_ALWAYS_SOFTWARE": "1"    // Works!
}, 
```
console output:
```
vscode@docker-desktop:/ros2_ws$ rviz2
[INFO] [1692057728.170884400] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1692057728.170984100] [rviz2]: OpenGl version: 4.5 (GLSL 4.5)
[INFO] [1692057728.202092000] [rviz2]: Stereo is NOT SUPPORTED
```

romain130492, 2024-08-25:
> @ygoumaz thank you for the reply. I agree with your conclusions here. Instead of forcing XQuartz to work properly on Apple Silicon, I ended up pursuing the software rendering route, using X11 [...]

Hey, thanks for making this. Were you by any chance also able to run gazebo?