ros2/rviz #935: Using RViz and Nav2 while displaying local cost map causes segmentation fault.

Zachaccino, 2022-12-29:
### System info

OS: Ubuntu 20.04
Nav2: 0.4.7-1focal.20221104.131617
RViz2: 8.2.7-1focal.20221013.020616

### Similar or related issues

https://github.com/ros2/rviz/issues/703 may have fixed the case for `tb3_simulation_launch.py` and I can verify that it would not crash when running this launch file. However, when I tried to do something similar to this launch file https://github.com/introlab/rtabmap_ros/blob/foxy-devel/launch/ros2/turtlebot3_rgbd.launch.py, I'm still experiencing segfault if I don't disable the visualization of local cost map.

I'm not entirely sure if this issue is identical to https://github.com/ros2/rviz/issues/799 but it sounds similar.

### Steps to reproduce 

The detailed instruction is shown in the comments of this launch file https://github.com/introlab/rtabmap_ros/blob/foxy-devel/launch/ros2/turtlebot3_rgbd.launch.py and following is the copypasta from the file. You will need RTABMAP and Turtlebot installed.

```
#   Install Turtlebot3 packages
#   Modify turtlebot3_waffle SDF:
#     1) Edit turtlebot3_gazebo/models/turtlebot3_waffle/model.sdf
#     2) Add
#          <joint name="camera_rgb_optical_joint" type="fixed">
#            <parent>camera_rgb_frame</parent>
#            <child>camera_rgb_optical_frame</child>
#            <pose>0 0 0 -1.57079632679 0 -1.57079632679</pose>
#            <axis>
#              <xyz>0 0 1</xyz>
#            </axis>
#          </joint> 
#     3) Rename <link name="camera_rgb_frame"> to <link name="camera_rgb_optical_frame">
#     4) Add <link name="camera_rgb_frame"/>
#     5) Change <sensor name="camera" type="camera"> to <sensor name="camera" type="depth">
#     6) Change image width/height from 1920x1080 to 640x480
#     7) Note that we can increase min scan range from 0.12 to 0.2 to avoid having scans 
#        hitting the robot itself
# Example:
#   $ export TURTLEBOT3_MODEL=waffle
#   $ ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
#
#   SLAM:
#   $ ros2 launch rtabmap_ros turtlebot3_rgbd.launch.py
#
#   Navigation (install nav2_bringup package):
#     $ ros2 launch nav2_bringup navigation_launch.py use_sim_time:=True
#     $ ros2 launch nav2_bringup rviz_launch.py
#
#   Teleop:
#     $ ros2 run turtlebot3_teleop teleop_keyboard
```

After running `ros2 launch nav2_bringup rviz_launch.py`, the following GUI would show up for 5-10 seconds before it crashes due to a segmentation fault and no additional errors were given in the log.

[image]

The output of RViz is shown below.
```
$ ros2 launch nav2_bringup rviz_launch.py
[INFO] [launch]: All log files can be found below /home/zac/.ros/log/2022-12-30-02-23-47-538778-haru-151501
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [151517]
[rviz2-1] [INFO] [1672327428.221598260] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1672327428.221738560] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-1] [INFO] [1672327428.305944118] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [WARN] [1672327428.401918183] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-1] [WARN] [1672327428.405693582] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-1] [WARN] [1672327428.408802457] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-1] [INFO] [1672327428.420237114] [rviz2]: Waiting for the lifecycle_manager_navigation/is_active service...
[rviz2-1] [INFO] [1672327428.420414525] [rviz2]: Sending lifecycle_manager_navigation/is_active request
[rviz2-1] [INFO] [1672327428.420720377] [rviz2]: Waiting for the lifecycle_manager_localization/is_active service...
[rviz2-1] [INFO] [1672327428.458686375] [rviz2]: Trying to create a map of size 145 x 138 using 1 swatches
[rviz2-1] [INFO] [1672327428.459506759] [rviz2]: Trying to create a map of size 60 x 60 using 1 swatches
[rviz2-1] [INFO] [1672327428.459867971] [rviz2]: Trying to create a map of size 145 x 138 using 1 swatches
[rviz2-1] [INFO] [1672327429.420848103] [rviz2]: Waiting for the lifecycle_manager_localization/is_active service...
[ERROR] [rviz2-1]: process has died [pid 151517, exit code -11, cmd '/opt/ros/foxy/lib/rviz2/rviz2 -d /opt/ros/foxy/share/nav2_bringup/rviz/nav2_default_view.rviz --ros-args -r __node:=rviz2'].

```

### Workaround

As mentioned in https://github.com/ros2/rviz/issues/703, disabling the visualization of `Controller` (or the local cost map) in the `Displays` panel would prevent this segmentation fault from happening.
  
I'm happy to help debugging this issue!

clalancette, 2023-01-02:
Yes, this is likely the same as what #703 fixes.  Unfortunately we can't backport that to Foxy, as it breaks ABI and API.  So we'd have to figure out a different solution for Foxy.

Zachaccino, 2023-01-08:
After some experimentation, showing the local cost map itself actually have no issues at all. The problem would only happen when I need to show other stuff like local plan or published footprint.

[image]

ARLunan, 2023-07-12:
I have encountered this same issue on rviz2 in Humble and the Segmentation Fault Occurs with the Local Cost Map enabled in rviz. Of course, a normal Nav2 rviz display needs to show Controller and Local Costmaps. So what is the status of a resolution for Humble and later distributions?

clalancette, 2023-07-13:
As far as we know, all of the issues here are fixed.  Please open up a new issue with detailed information about your system and setup, and a backtrace of the crash occurring.

ARLunan, 2023-07-13:
Thank you for your response and the status of this issue #935. The reason for my post was initiated by a message from Parallels Support, pertaining to their MAC Virtual Machine application that I use to run Ubuntu 22.04 Full Desktop / ROS2 Humble on my MAC M1, which is a arm64 / aarch64 CPU/GPU.  Their response to me was to revert back to ROS2 on on my Issue: Rviz2 that is displaying Nav2 Costmaps (A segmentation fault ?) - crashing when then VM "3D Access" feature is enabled. 

rviz2 works with 3D Accel disabled. While the Ubuntu "tops" CPU load is high (~95%) when rviz2 is launched, the Nav2 system is functional. This Parallels Desktop Ver 18 introduced support for VirGL allowing for hardware acceleration for hardware graphics acceleration in Linux virtual machines. I confirmed that Virgl is enabled whether 3D Accel is enabled or not. 

When I run rviz2 w/o 3D Accelleration enabled, this is printed: Warning : Ignoring XDG_SESSION_TYPE=wayland on Gnome Use QT_QPA_PLATFORM to run on Wayland anyway. 

When I run rviz2 w/ 3D Acell enabled, this is printed: Warning : Ignoring XDG_SESSION_TYPE=wayland on Gnome Use QT_QPA_PLATFORM to run on Wayland anyway. 
libGL error: pci id for fd 40: lab8:0010, driver, null   
pci id for fd 41: lab8:0010, driver, null 
libGL error: pci id for fd 44: lab8:0010, driver, null   
pci id for fd 45: lab8:0010, driver, null 

So unless you advise otherwise, based on the status of this issue #935, there appears to be no action required by Ros2 Tech Support & I should get back to Parallels Support for action.

clalancette, 2023-07-14:
Thanks for the follow up.  I'm actually going to close this issue out for now, but if you do think that it is an issue in ROS 2 please feel free to open a new ticket with details.

closed 2023-07-14 by clalancette