ros2/rviz #937: Constant texture recreation for costmap

Eric-Bwr, 2023-01-04:
The texture is just being recreated instead of updated. 
One could use something like this:

`auto buffer = texture_->getBuffer();
            buffer->lock(Ogre::HardwareBuffer::HBL_DISCARD);
            const Ogre::PixelBox &pb = buffer->getCurrentLock();
            auto *data = static_cast<uint8_t*>(pb.data);
            size_t pitch = pb.rowPitch;
            for(uint32_t y = 0; y < map.info.height; ++y){
                for(uint32_t x = 0; x < map.info.width; ++x){
                    data[pitch * y + x] = SOMEDATA;
                }
            }
            buffer->unlock();`

clalancette, 2023-01-04:
Sorry, this isn't enough information to figure out what you mean.  Can you give an example of a problem you are seeing, what platform you are using, the ROS distribution you are using, and a better description of what changes you think will help?

Eric-Bwr, 2023-01-04:
Oh sorry, I needed to head to a meeting so I forgot about this :/ It should be clear now :)