ros2/rviz #940: Embed RViz2 into a custom QT C++ gui as a dock widget

rebeccaRossRobotics, 2023-01-19:
Hi,

I'm currently trying to get an RViz view into a custom QT C++ GUI as a dock widget. I followed the example from this [repository](https://github.com/mjeronimo/rviz_embed_test) and have managed to get RViz into a QWidget and is working as expected. When I put that widget into a QDockWidget everything works as expected until I move that dock widget on top of another dock widget. Below shows some pictures of the issue for context.
[image]
[image]

This [repo](https://github.com/rebeccaRossRobotics/qt_functionality_testing.git) contains my QT project and code. 
To reproduce the issue, go to the gui_ws fold, build everything with `colcon build`, source the workspace and then run with `ros2 run gui gui`. Once open drag the RViz dock on top of one of the other docks.

Does anyone know why this is happening? And if possible how to fix it?

Thanks
Rebecca

whudesk, 2023-06-15:
myviz.cpp.obj:-1: error: LNK2019: 无法解析的外部符号 "public: __cdecl rviz_common::VisualizationManager::VisualizationManager(class rviz_common::RenderPanel *,class std::weak_ptr<class rviz_common::ros_integration::RosNodeAbstractionIface>,class rviz_common::WindowManagerInterface *,class std::shared_ptr<class rclcpp::Clock>)" (??0VisualizationManager@rviz_common@@QEAA@PEAVRenderPanel@1@V?$weak_ptr@VRosNodeAbstractionIface@ros_integration@rviz_common@@@std@@PEAVWindowManagerInterface@1@V?$shared_ptr@VClock@rclcpp@@@4@@Z)，函数 "private: void __cdecl MyViz::initializeRViz(void)" (?initializeRViz@MyViz@@AEAAXXZ) 中引用了该符号

peter5232, 2023-07-03:
> Hi,
>
> I'm currently trying to get an RViz view into a custom QT C++ GUI as a dock widget. I followed the example from this [repository](https://github.com/mjeronimo/rviz_embed_test) and have managed [...]
I tried this repo (https://github.com/mjeronimo/rviz_embed_test).He can compile, but the rviz embedding module inside does not support mouse operations (zoom in, zoom out, rotate). I have checked the rviz code in rviz box and the function call order is consistent with that in this repo. Can you provide a solution?

Tjrandhawa26, 2024-05-13:
> I tried this repo (https://github.com/mjeronimo/rviz_embed_test).He can compile, but the rviz embedding module inside does not support mouse operations (zoom in, zoom out, rotate). I have checked the rviz code in rviz box and the function call order is consistent with that in this repo. Can you provide a solution?
> 
> > Hi,
> > I'm currently trying to get an RViz view into a custom QT C++ GUI as a dock widget. I followed the example from this [repository](https://github.com/mjeronimo/rviz_embed_test) and have managed to get RViz into a QWidget and is working as expected. When I put that widget into a QDockWidget everything works as expected until I move that dock widget on top of another dock widget. Below shows some pictures of the issue for context. [image] [image]
> > This [repo](https://github.com/rebeccaRossRobotics/qt_functionality_testing.git) contains my QT project and code. To reproduce the issue, go to the gui_ws fold, build everything with `colcon build`, source the workspace and then run with `ros2 run gui gui`. Once open drag the RViz dock on top of one of the other docks.
> > Does anyone know why this is happening? And if possible how to fix it?
> > Thanks Rebecca

did you get the solution for this?

XieXinJian, 2024-09-09:
rviz_common::ToolManager* toolManager = manager_->getToolManager();
    //the default plugins init error by default code.need to change source code
    auto plugins = toolManager->getFactory()->getDeclaredPlugins();
    for(auto plugin:plugins)
    {
        toolManager->addTool(plugin);
    }
    toolManager->setCurrentTool(toolManager->getTool(1));
    toolManager->setDefaultTool(toolManager->getTool(1));
    
   should add tool plugin by self , because the manager initialize function failed to add this tools. i thought this is because of the code was't update. my brach is humble,and there also have many other errors can't resolve now