ros2/rviz #948: Black Screen on version 11.2.5

neutronztar, 2023-02-01:
When I open Rviz2 with the command 'Rviz2' I get a black screen where the grid and models usually are. This was working on 11.2.4.

OS: Windows 10 host, Rviz2 running a WSL Docker container

Note: My friend is running an Ubuntu host with a Docker container and having the same problem with 11.2.5.

[image]

clalancette, 2023-02-01:
Hm, I can't reproduce this here; it seems to work fine for me in a container (though on Ubuntu, not WSL).

The only change between 11.2.4 and 11.2.5 of consequence was #931 .  While I guess that could cause problems, it seems unlikely that this is causing this particular problem.

If you downgrade *just* rviz_rendering by doing:

```
$ wget http://snapshots.ros.org/humble/2022-12-16/ubuntu/pool/main/r/ros-humble-rviz-rendering/ros-humble-rviz-rendering_11.2.4-1jammy.20221108.154346_amd64.deb
$ sudo dpkg -i ros-humble-rviz-rendering_11.2.4-1jammy.20221108.154346_amd64.deb
```

Does that make it work for you?

neutronztar, 2023-02-01:
I'm having some trouble downgrading with that command, after running those and opening rviz I'm still on 11.2.5. Here are the results of 'apt list -- installed | grep rviz'

`ros-humble-rviz-assimp-vendor/now 11.2.5-1jammy.20230112.141217 amd64 [installed,local]
ros-humble-rviz-common/now 11.2.5-1jammy.20230120.222053 amd64 [installed,local]
ros-humble-rviz-default-plugins/now 11.2.5-1jammy.20230124.071257 amd64 [installed,local]
ros-humble-rviz-ogre-vendor/now 11.2.5-1jammy.20230112.141237 amd64 [installed,local]
ros-humble-rviz-rendering/now 11.2.4-1jammy.20221108.154346 amd64 [installed,local]
ros-humble-rviz2/now 11.2.5-1jammy.20230124.081000 amd64 [installed,local]`

Doesn't seem like this downgraded all of rviz.

Which version of Ubuntu are you on? My docker container is 22.0.4. It is working when I install it directly in my WSL distro which is Ubuntu 20.04.

clalancette, 2023-02-01:
> Doesn't seem like this downgraded all of rviz.

Yeah, that's correct.  The command I gave you only downgraded rviz_rendering, which is the only thing that changed between 11.2.4 and 11.2.5.  It should still run.

> Which version of Ubuntu are you on?

Ubuntu 22.04, running in an LXC container.

VRichardJP, 2023-02-02:
I am facing a black screen issue aswell, using a docker container on ubuntu.

I initially had rviz2 11.2.5 installed:
```
$ sudo apt list --installed | grep rviz

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

ros-humble-rviz-assimp-vendor/jammy,now 11.2.5-1jammy.20230112.141217 amd64 [installed,automatic]
ros-humble-rviz-common/jammy,now 11.2.5-1jammy.20230120.222053 amd64 [installed,automatic]
ros-humble-rviz-default-plugins/jammy,now 11.2.5-1jammy.20230124.071257 amd64 [installed,automatic]
ros-humble-rviz-ogre-vendor/jammy,now 11.2.5-1jammy.20230112.141237 amd64 [installed,automatic]
ros-humble-rviz-rendering/jammy,now 11.2.5-1jammy.20230112.145716 amd64 [installed,automatic]
ros-humble-rviz2/jammy,now 11.2.5-1jammy.20230124.081000 amd64 [installed,automatic]
```

Then I tried to downgrade everything to 11.2.4
```
$ sudo apt list --installed | grep rviz

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

ros-humble-rviz-assimp-vendor/now 11.2.4-1jammy.20221108.145620 amd64 [installed,upgradable to: 11.2.5-1jammy.20230112.141217]
ros-humble-rviz-common/now 11.2.4-1jammy.20221207.043657 amd64 [installed,upgradable to: 11.2.5-1jammy.20230120.222053]
ros-humble-rviz-default-plugins/now 11.2.4-1jammy.20221207.063337 amd64 [installed,upgradable to: 11.2.5-1jammy.20230124.071257]
ros-humble-rviz-ogre-vendor/now 11.2.4-1jammy.20221108.145633 amd64 [installed,upgradable to: 11.2.5-1jammy.20230112.141237]
ros-humble-rviz-rendering/now 11.2.4-1jammy.20221108.154346 amd64 [installed,upgradable to: 11.2.5-1jammy.20230112.145716]
ros-humble-rviz2/now 11.2.4-1jammy.20221207.075810 amd64 [installed,upgradable to: 11.2.5-1jammy.20230124.081000]
```
-> Black screen

Then to 11.2.3:
```
$ sudo apt list --installed | grep rviz

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

ros-humble-rviz-assimp-vendor/now 11.2.3-1jammy.20220913.012626 amd64 [installed,upgradable to: 11.2.5-1jammy.20230112.141217]
ros-humble-rviz-common/now 11.2.3-1jammy.20221019.162459 amd64 [installed,upgradable to: 11.2.5-1jammy.20230120.222053]
ros-humble-rviz-default-plugins/now 11.2.3-1jammy.20221019.165603 amd64 [installed,upgradable to: 11.2.5-1jammy.20230124.071257]
ros-humble-rviz-ogre-vendor/now 11.2.3-1jammy.20220913.012843 amd64 [installed,upgradable to: 11.2.5-1jammy.20230112.141237]
ros-humble-rviz-rendering/now 11.2.3-1jammy.20220913.020348 amd64 [installed,upgradable to: 11.2.5-1jammy.20230112.145716]
ros-humble-rviz2/now 11.2.3-1jammy.20221019.180719 amd64 [installed,upgradable to: 11.2.5-1jammy.20230124.081000]
```
-> Black screen

Then to 11.2.2:
```
$ sudo apt list --installed | grep rviz

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

ros-humble-rviz-assimp-vendor/now 11.2.2-1jammy.20220520.000232 amd64 [installed,upgradable to: 11.2.5-1jammy.20230112.141217]
ros-humble-rviz-common/now 11.2.2-1jammy.20220908.235220 amd64 [installed,upgradable to: 11.2.5-1jammy.20230120.222053]
ros-humble-rviz-default-plugins/now 11.2.2-1jammy.20220909.025059 amd64 [installed,upgradable to: 11.2.5-1jammy.20230124.071257]
ros-humble-rviz-imu-plugin/now 2.1.1-2jammy.20220909.002940 amd64 [installed,upgradable to: 2.1.3-1jammy.20230120.224430]
ros-humble-rviz-ogre-vendor/now 11.2.2-1jammy.20220520.000242 amd64 [installed,upgradable to: 11.2.5-1jammy.20230112.141237]
ros-humble-rviz-rendering/now 11.2.2-1jammy.20220520.003608 amd64 [installed,upgradable to: 11.2.5-1jammy.20230112.145716]
ros-humble-rviz2/now 11.2.2-1jammy.20220909.043241 amd64 [installed,upgradable to: 11.2.5-1jammy.20230124.081000]
```
-> still black screen

There is no version earlier than 11.2.2 on Humble. But I have used Rviz in humble containers before, and it was working just fine. So it does not seem to be a Rviz2 version issue (in my case at least).

VRichardJP, 2023-02-02:
Simple test with rocker:

- on foxy:
```
$ rocker --x11 osrf/ros:foxy-desktop
# rviz2
```
-> the default grid is visible

- on humble:
```
$ rocker --x11 osrf/ros:humble-desktop
# rviz2
```
-> the 3D view is all black

The images I pulled:
```
docker images | grep osrf  148ms  Thu 02 Feb 2023 12:29:05 PM JST
osrf/ros                                                       foxy-desktop                  0630734c5305   3 hours ago         3.12GB
osrf/ros                                                       humble-desktop                734e74e8d052   27 hours ago        3.57GB
```

@neutronztar could you test rviz from `osrf/ros:humble-desktop` on your machine?

clalancette, 2023-02-02:
Can you also try to remove the directory in `$HOME/.rviz2` to see if that improves things at all?

VRichardJP, 2023-02-03:
@clalancette removing `$HOME/.rviz2` seems not to have any impact.

VRichardJP, 2023-02-03:
I observed something interesting though:
In the fovy container, OpenGL 3.1 is used:
```
root@2ed4539840e3:/# rviz2
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[INFO] [1675390752.635322008] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1675390752.635524874] [rviz2]: OpenGl version: 3.1 (GLSL 1.4)
[INFO] [1675390752.833427792] [rviz2]: Stereo is NOT SUPPORTED
```

from the humble one, it is OpenGL 4.5:
```
root@ec83563c233e:/# rviz2
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[INFO] [1675389782.489131125] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1675389782.489196376] [rviz2]: OpenGl version: 4.5 (GLSL 4.5)
[INFO] [1675389782.529771343] [rviz2]: Stereo is NOT SUPPORTED
```

What if it is a problem with OpenGL? I have tested `glxgears` from `mesa-utils` package in both container:
- In the foxy one, the demo runs normally
- In the humble one, I get the following error:
```
root@9d465a06b92b:/# glxgears
X Error of failed request:  BadShmSeg (invalid shared segment parameter)
  Major opcode of failed request:  130 (MIT-SHM)
  Minor opcode of failed request:  3 (X_ShmPutImage)
  Segment id in failed request:  0x3400005
  Serial number of failed request:  43
  Current serial number in output stream:  44
```

JensVanhooydonck, 2023-02-04:
When using docker adding `---ipc=host` to the run command or adding `ipc: host` to the docker-compose file fixes this for me.

VRichardJP, 2023-02-06:
@JensVanhooydonck 
It does not seem to do the trick on my machine. Here for example a simple docker file to run rviz2 on foxy/humble:
```
version: "3"
services:
  rviz2_test:
    image: osrf/ros:foxy-desktop
    # image: osrf/ros:humble-desktop
    command: rviz2
    environment:
      - DISPLAY
      - XAUTHORITY=/tmp/.Xauthority
    volumes: 
      - ~/.Xauthority:/tmp/.Xauthority
      - /tmp/.X11-unix:/tmp/.X11-unix
    network_mode: host
```

- With the foxy image, Rviz2 shows the grid.
- With the humble image, Rviz2 shows a black view.
- With the humble image and `ipc: host`, Rviz2 still shows a black display

VRichardJP, 2023-02-06:
There is really a thing between ubuntu 20.04 and 22.04:

- Running `glxgears` fails with `ubuntu:22.04`:
```
$ docker run --rm --network=host -e DISPLAY -e XAUTHORITY=/tmp/.Xauthority -v ~/.Xauthority:/tmp/.Xauthority -v /tmp/.X11-unix:/tmp/.X11-unix ubuntu:22.04 /bin/bash -c "apt update && apt install -y mesa-utils && glxgears"
...
Setting up libgl1:amd64 (1.4.0-1) ...
Setting up mesa-utils-bin:amd64 (8.4.0-1ubuntu1) ...
Setting up mesa-utils (8.4.0-1ubuntu1) ...
Processing triggers for libc-bin (2.35-0ubuntu3.1) ...
X Error of failed request:  BadShmSeg (invalid shared segment parameter)
  Major opcode of failed request:  130 (MIT-SHM)
  Minor opcode of failed request:  3 (X_ShmPutImage)
  Segment id in failed request:  0x4600005
  Serial number of failed request:  43
  Current serial number in output stream:  44
```

But no issue with `ubuntu:20.04`:
```
$ docker run --rm --network=host -e DISPLAY -e XAUTHORITY=/tmp/.Xauthority -v ~/.Xauthority:/tmp/.Xauthority -v /tmp/.X11-unix:/tmp/.X11-unix ubuntu:20.04 /bin/bash -c "apt update && apt install -y mesa-utils && glxgears"
```

[image]

JensVanhooydonck, 2023-02-06:
I've also added this to the environment variables:
      - MESA_GL_VERSION_OVERRIDE=3.1COMPAT
      - MESA_GLES_VERSION_OVERRIDE=3.1
To match the OpenGL version i run locally (In a VM on a M1 macbook)

tinducvo, 2023-02-06:
I also have this error, but only on 1/2 machines with identical rviz2 versions
```This is RViz version 11.2.5 (ROS 2).

Compiled against Qt version 5.15.3.
Compiled against OGRE version 1.12.1 (Rhagorthua)```

tinducvo, 2023-02-06:
I am using this docker image: https://hub.docker.com/r/tiryoh/ros2-desktop-vnc/tags?page=1&name=humble.
Everything seems to be working after downgrading to tag humble-20230129T1423.
The ros-humble-desktop versions were both ros-humble-desktop/jammy,now 0.10.0-1jammy.20230124.081312 amd64.
Let me know if you want me to test anything.

hanjianwei92, 2023-02-12:
I also encountered this problem. everything that render 3D  display black screen, include rviz  gazebo and moveit_setup assisant.
This happens when I use docker image provided by moveit and new ubuntu22.04 image  manually installed ros and moveit.

Stefan-working, 2023-02-13:
I also had this problem with windows 10 enterprise, docker-desktop and Ubuntu 22.04 LTS. I was able to fix this issue by upgrading the mesa package with the binaries provided by the kisak/kisak-mesa ppa:

> add-apt-repository ppa:kisak/kisak-mesa 
> apt update
> apt upgrade

DatSpace, 2023-02-13:
I can verify the issue in docker with an image based on the latest ubuntu22.04 and ros2 humble from the latest apt packages.

VRichardJP, 2023-02-14:
I confirm the mesa upgrade from `ppa:kisak/kisak-mesa` does the trick:

- glxgears now works:
```
docker run --rm --network=host -e DISPLAY -e XAUTHORITY=/tmp/.Xauthority -v ~/.Xauthority:/tmp/.Xauthority -v /tmp/.X11-unix:/tmp/.X11-unix ubuntu:22.04 /bin/bash -c "apt update && apt install -y software-properties-common && add-apt-repository ppa:kisak/kisak-mesa && apt install -y mesa-utils && glxgears"
```

- same for rviz2 on humble:
```
docker run --rm --network=host -e DISPLAY -e XAUTHORITY=/tmp/.Xauthority -v ~/.Xauthority:/tmp/.Xauthority -v /tmp/.X11-unix:/tmp/.X11-unix osrf/ros:humble-desktop /bin/bash -c "apt update && apt install -y software-properties-common && add-apt-repository ppa:kisak/kisak-mesa && apt upgrade -y && rviz2"
```

On my machine the `apt upgrade` installed:
```
libegl-mesa0 libegl1-mesa-dev libgbm-dev libgbm1 libgl1-mesa-dev
  libgl1-mesa-dri libglapi-mesa libglx-mesa0 libllvm15 libpam-modules
  libpam-modules-bin libpam-runtime libpam0g libssl-dev libssl3 linux-libc-dev
  openssl python3-colcon-core python3-colcon-defaults python3-colcon-mixin
  python3-colcon-python-setup-py
```

- there are more than just a few mesa libs, so I also tested installing just the mesa related packages. This one works too:
```
docker run --rm --network=host -e DISPLAY -e XAUTHORITY=/tmp/.Xauthority -v ~/.Xauthority:/tmp/.Xauthority -v /tmp/.X11-unix:/tmp/.X11-unix osrf/ros:humble-desktop /bin/bash -c "apt update && apt install -y software-properties-common && add-apt-repository ppa:kisak/kisak-mesa && apt install libegl-mesa0 libegl1-mesa-dev libgbm-dev libgbm1 libgl1-mesa-dev libgl1-mesa-dri libglapi-mesa libglx-mesa0 -y && rviz2"
```

- without the packages from kisak-mesa, rviz2 3D view is black:
```
docker run --rm --network=host -e DISPLAY -e XAUTHORITY=/tmp/.Xauthority -v ~/.Xauthority:/tmp/.Xauthority -v /tmp/.X11-unix:/tmp/.X11-unix osrf/ros:humble-desktop /bin/bash -c "apt update && apt upgrade -y && rviz2"
```

@Stefan-working Thank you for the fix!
Hopefully, the mesa available in ubuntu 22.04 will be upgraded soon enough for the problem to disappear itself.

Cheny5863, 2023-03-20:
> I also had this problem with windows 10 enterprise, docker-desktop and Ubuntu 22.04 LTS. I was able to fix this issue by upgrading the mesa package with the binaries provided by the kisak/kisak-mesa ppa:
> 
> > add-apt-repository ppa:kisak/kisak-mesa
> > apt update
> > apt upgrade

it works!!! thank u very much!

Cheny5863, 2023-03-20:
I also had this problem with Ubuntu 18.04LTS, docker and Ubuntu 22.04 LTS. I was able to fix this issue by add option "--privileged" when run container，like this:
`sudo docker run -it -v /tmp/.X11-unix -e DISPLAY=unix$DISPLAY -e GDK_SCALE -e GDK_DPI_SCALE --net=host --privileged osrf/ros:humble-desktop-full`

swiz23, 2023-03-30:
I've been able to get around this by just installing nvidia-driver-515 in my docker image. 
This installed the following packages:
```
  cpp-12 dctrl-tools dkms gcc-12 keyboard-configuration kmod libasan8 libgcc-12-dev liblocale-gettext-perl libnvidia-cfg1-515 libnvidia-common-515 libnvidia-compute-515 libnvidia-decode-515 libnvidia-egl-wayland1 libnvidia-encode-515 libnvidia-extra-515 libnvidia-fbc1-515
  libnvidia-gl-515 libtsan2 libxcvt0 nvidia-compute-utils-515 nvidia-dkms-515 nvidia-driver-515 nvidia-kernel-common-515 nvidia-kernel-source-515 nvidia-utils-515 udev xserver-xorg-core xserver-xorg-video-nvidia-515
```

clalancette, 2023-03-30:
There is a lot of information in here, but everything in this issue points to user configuration, not problems in the code.  Since that seems to be the case, I'm going to close this issue out; for further issues on getting RViz to run, please use https://answers.ros.org .

closed 2023-03-30 by clalancette

elms, 2023-04-28:
I've been running into this and finally spent some time digging. It appears to be a known issue in ubuntu with the update to mesa, see [bug report](https://bugs.launchpad.net/ubuntu/+source/mesa/+bug/2004649).

I also found this [PR](https://github.com/RobotLocomotion/drake-ci/pull/217/files) helpful  for how to downgrade and make sure the libraries aren't upgraded again by accident. For my specific case I also had to downgrade `libgbm-dev=22.0.1-1ubuntu2 libsdl2-2.0-0=2.0.20+dfsg-2build1 libsdl2-dev=2.0.20+dfsg-2build1` and hold them.

rubenanapu, 2023-06-09:
The solution found by @Stefan-working worked for me too (Thanks Stefan):

> I also had this problem with windows 10 enterprise, docker-desktop and Ubuntu 22.04 LTS. I was able to fix this issue by upgrading the mesa package with the binaries provided by the kisak/kisak-mesa ppa:
> 
> > add-apt-repository ppa:kisak/kisak-mesa
> > apt update
> > apt upgrade

hodnajit, 2023-07-27:
> I confirm the mesa upgrade from `ppa:kisak/kisak-mesa` does the trick:
> 
>     * glxgears now works:
> 
> 
> ```
> docker run --rm --network=host -e DISPLAY -e XAUTHORITY=/tmp/.Xauthority -v ~/.Xauthority:/tmp/.Xauthority -v /tmp/.X11-unix:/tmp/.X11-unix ubuntu:22.04 /bin/bash -c "apt update && apt install -y software-properties-common && add-apt-repository ppa:kisak/kisak-mesa && apt install -y mesa-utils && glxgears"
> ```
> 
>     * same for rviz2 on humble:
> 
> 
> ```
> docker run --rm --network=host -e DISPLAY -e XAUTHORITY=/tmp/.Xauthority -v ~/.Xauthority:/tmp/.Xauthority -v /tmp/.X11-unix:/tmp/.X11-unix osrf/ros:humble-desktop /bin/bash -c "apt update && apt install -y software-properties-common && add-apt-repository ppa:kisak/kisak-mesa && apt upgrade -y && rviz2"
> ```
> 
> On my machine the `apt upgrade` upgraded:
> 
> ```
> libegl-mesa0 libegl1-mesa-dev libgbm-dev libgbm1 libgl1-mesa-dev
>   libgl1-mesa-dri libglapi-mesa libglx-mesa0 libllvm15 libpam-modules
>   libpam-modules-bin libpam-runtime libpam0g libssl-dev libssl3 linux-libc-dev
>   openssl python3-colcon-core python3-colcon-defaults python3-colcon-mixin
>   python3-colcon-python-setup-py
> ```
> 
>     * there are more than just a few mesa libs, so I also tested installing just the mesa related packages. This one works too:
> 
> 
> ```
> docker run --rm --network=host -e DISPLAY -e XAUTHORITY=/tmp/.Xauthority -v ~/.Xauthority:/tmp/.Xauthority -v /tmp/.X11-unix:/tmp/.X11-unix osrf/ros:humble-desktop /bin/bash -c "apt update && apt install -y software-properties-common && add-apt-repository ppa:kisak/kisak-mesa && apt install libegl-mesa0 libegl1-mesa-dev libgbm-dev libgbm1 libgl1-mesa-dev libgl1-mesa-dri libglapi-mesa libglx-mesa0 -y && rviz2"
> ```
> 
>     * without the packages from kisak-mesa, rviz2 3D view is black:
> 
> 
> ```
> docker run --rm --network=host -e DISPLAY -e XAUTHORITY=/tmp/.Xauthority -v ~/.Xauthority:/tmp/.Xauthority -v /tmp/.X11-unix:/tmp/.X11-unix osrf/ros:humble-desktop /bin/bash -c "apt update && apt upgrade -y && rviz2"
> ```
> 
> @Stefan-working Thank you for the fix! Hopefully, the mesa available in ubuntu 22.04 will be upgraded soon enough for the problem to disappear itself. It seems the latest mesa available on Ubuntu 22.04 is 22.0.1: https://packages.ubuntu.com/search?suite=jammy&arch=any&searchon=names&keywords=mesa With the kisak repo I could upgrade to version 22.3.5

Thanks! That fixes it for me