ros2/rviz #951: Bug - Rviz GLSL

nadavis, 2023-02-17:
Ubuntu 22.04
Ros2 Humble

Hi, Many thanks for the framework very clear, it helped me alot understanding SLAM.
unfortently i keep getting a error once i change the Slam Toolbox to work with localization, a saved map is not loading, i am keeping getting a new map visualization, moreover, after i changed RViz to work with map i get the follow error, please any idea, i did not find hint in Google
"
Vertex Program:rviz/glsl120/indexed_8bit_image.vert Fragment Program:rviz/glsl120/indexed_8bit_image.frag GLSL link result :
[rviz2-1] active samplers with a different type refer to the same texture image unit
"

clalancette, 2023-02-17:
Please don't open up the same issue on multiple repositories; it causes us to have to do more work to track and close them.  Closing this one.

closed 2023-02-17 by clalancette

nadavis, 2023-02-18:
Ok, many thanks!!
Sorry for the trouble..