ros2/rviz #956: Using service client to get data in rviz2 plguin

nrasulnrasul, 2023-03-08:
In a custom rviz2 plugin, I try to display data received by client from server.

I got the following error.

```
terminate called after throwing an instance of 'std::runtime_error'
what():  spin_once() called while already spinning
```
My code is similar to the following

[https://github.com/introlab/rtabmap_ros/blob/humble-devel/src/rviz/MapCloudDisplay.cpp#L557](https://github.com/introlab/rtabmap_ros/blob/humble-devel/src/rviz/MapCloudDisplay.cpp#L557)

Is it possible at the moment to use client in rviz plugin to get receive data.

clalancette, 2023-03-08:
Please open questions like this on https://answers.ros.org, which is our central Question and Answer site.  You'll get a better answer there, and it will be searchable for the future.

Make sure to include a lot of information on what platform you are using, which ROS distribution you are using, and the exact steps you took.

closed 2023-03-08 by clalancette

nrasulnrasul, 2023-03-08:
Did you even read the question ? and look the code in link ?