ros2/rviz #957: Variance field for the Range message

ejalaa12, 2023-03-13:
The `Range` message will get an additional field for the variance.
I don't know if this is relevant for you guys, but this issue is simply a notification, feel free to close it if it's not a change that you could benefit from.

Although for RViz, I see a potential new feature to display the range variance.

You can see details about the PR [here](https://github.com/ros2/common_interfaces/pull/181)

clalancette, 2023-03-23:
It would be a good enhancement to RViz to actually look at and display this data.  So I'll leave this open on the backlog.

bryangd34, 2023-06-17:
Has this feature been implemented?
I am looking to start contributing to the ROS 2 project. This seems like a **good first issue** :)

ejalaa12, 2023-06-18:
Hi @bryangd34 I don't think this has been implemented yet.
This is indeed a good first issue.
 How would you display the variance in rviz?

bryangd34, 2023-06-18:
Hi @ejalaa12 
Right now, I cannot say how I will display the variance in rviz. I would need to start looking into display options and then start working on implementing this. 
First, I want to be sure there is still a need for this feature.

ejalaa12, 2023-06-18:
The range sensor uses a cone to display the range value. One way I imagined displaying the variance was to have a slice of the cone colored a different color representing the variance.

But I don't know if this representation actually brings something

bryangd34, 2023-06-20:
That's a good idea, the slice. 
What about, more generally, other message types. How do they show variance? Perhaps there is some inspiration there. I'll investigate.

ejalaa12, 2023-06-21:
The only Rviz Display that show variance is the PoseWithCovariance or Odometry. 
It basically draws the 3D covariance ellipse error.

Randil-Fernando, 2026-10-09:
Hey, can I take this one?

Had a look and `variance` did land in `sensor_msgs/Range` (ros2/common_interfaces#181), it's in rolling now. `range_display.cpp` still ignores it though, cone's built from `range` + `field_of_view` only.

My idea: take `sigma = sqrt(variance)` and draw a faded second cone from `range - sigma` to `range + sigma`, so you get a band around the normal one. Skip it when `variance == 0` since the msg says that means unknown. Put it behind a `Show Variance` checkbox, off by default, so nothing changes unless you turn it on.

Or I could just tint a slice of the existing cone like suggested above, don't mind either way.

Sound good? I'll start once someone confirms the look

bryangd34, 2026-10-09:
@Randil-Fernando yea fine by me. Things become pretty busy for me around this time so never made much progress. I think your idea is great :) very curious to see how you get on.