ros2/rviz #966: black backgroud splash ?

chopin1998, 2023-04-04:
my os is ubuntu 22.04, upgraded to latest.

recently, my rviz2 background is splashing...even there is nothing item was selected.

and no error log from terminal,

what's the problem?

https://user-images.githubusercontent.com/643792/229682354-a14a7714-604d-4c7a-b602-509020994b06.mp4

chopin1998, 2023-04-06:
i got a temp solution... when the background flash, i switch to other desktop (Win+F1/F2/F3....), the switch back

it display ok, but you cannot resize the window, otherwise flash again...

i have no idea now...

Yunhuang-Liu, 2023-07-21:
I also get this bug, ros humbel(Ubuntu22.04).

CatInTheRain, 2023-07-23:
I found the same error while i was using a docker. Studying the dockerfile. I tried to remove the following installed package:
```bash
sudo apt uninstall qt5dxcb-plugin
```
and I solve this issue. I don't know if it could be usefull

Yunhuang-Liu, 2023-07-24:
> I found the same error while i was using a docker. Studying the dockerfile. I tried to remove the following installed package:
> 
> ```shell
> sudo apt remove qt5dxcb-plugin
> ```
> 
> and I solve this issue. I don't know if it could be usefull

Nice, it is useful.Very thanks for you.

chopin1998, 2023-07-24:
> 

c00l, but my other software depends on qt5dxcb-plugin......