ros2/rviz #974: `MovableText::getWorldTransforms` only applies global translation in y-Direction

gkueppers, 2023-04-24:
Hello all,

I don't understand the current implementation of the `MovableText::getWorldTransforms` function. In line 521, the `global_translation_` is multiplied by Ogre::Vector3::UNIT_Y. The `global_translation_` is therefore only applied in the y-direction. Is this a deliberate choice or is this a bug?

https://github.com/ros2/rviz/blob/1ac419472ed06cdd52842a8f964f953a75395245/rviz_rendering/src/rviz_rendering/objects/movable_text.cpp#L521

I also looked at the ROS1 implementation and here [the whole translation is added](https://github.com/ros-visualization/rviz/blob/ec7ab1b0183244c05fbd2d0d1b8d8f53d8f42f2b/src/rviz/ogre_helpers/movable_text.cpp#L506).

Perhaps someone can give me an opinion here or explain the intention behind this, which I currently don't understand!

Best regards
Guido

clalancette, 2023-05-04:
If you wouldn't mind opening a PR, backporting the change from ROS 1, we're happy to take a look and review it.  Thanks in advance.

yaswanth1701, 2024-01-05:
Hi @clalancette I would like to work on this issue. Can I ?

clalancette, 2024-01-05:
> Hi @clalancette I would like to work on this issue. Can I ?

Please feel free!

PR #1118 (2024-01-10): Fixed the global translation in `MovableText::getWorldTransforms `
closes #974 
Back-ported changes from ROS 1 for  `MovableText::getWorldTransforms ` which was previously applying global translation only in y direction.
closed 2024-01-10 by ahcorde