ros2/rviz #980: Suggest to changeAccess specifier in PoseWithCovarianceDisplay class

hyunseok-yang, 2023-05-10:
Hi, 

I'm trying to make a new type of Rviz Display plugin class which is derivied from rviz_default_plugins::displays::PoseWithCovarianceDisplay class.

But as we can see below line, I cannot override the function since it is private virtual function.
The function's name is processMessage().

https://github.com/ros2/rviz/blob/82538e5b4453b1b85f0b49a206d8b225f65b2424/rviz_default_plugins/include/rviz_default_plugins/displays/pose_covariance/pose_with_covariance_display.hpp#L103

As far as I know, most of this function in each dispaly plugin class are written as protected access specifier.

Could we modify to 'protected:'??

hyunseok-yang, 2023-05-10:
I requested PR https://github.com/ros2/rviz/pull/981

Yadunund, 2023-05-10:
Looks like there is some inconsistency across various [display plugins](https://github.com/ros2/rviz/tree/rolling/rviz_default_plugins/include/rviz_default_plugins/displays) with regards to the scope of `processMessage()` so it might be good to standardize across all of them in the same effort.

In the meanwhile, another approach could be to write your own plugin that internally instantiates an object of the existing `PoseWithCovarianceDisplay` type. Then implement the new plugin methods such that they simply calls the respective methods of this object except for `processMessage()` which you implement with custom logic.

hyunseok-yang, 2023-05-10:
> Looks like there is some inconsistency across various [display plugins](https://github.com/ros2/rviz/tree/rolling/rviz_default_plugins/include/rviz_default_plugins/displays) with regards to the [...]

Thanks for you comment :)
I'm digging other way instead just for temporarily.

hyunseok-yang, 2023-05-10:
@Yadunund  
Yadu, it looks there was some issue on jenkin though build result looks good(?). 
https://build.ros2.org/job/Hpr__rviz__ubuntu_jammy_amd64/19/

Do you know why it failed? :(

hyunseok-yang, 2023-05-11:
According to log(https://build.ros2.org/job/Hpr__rviz__ubuntu_jammy_amd64/19/console), code style? and cpplint check may failed.

closed 2023-05-17 by clalancette