ros2/rviz #986: How can i solve it? (Failed   <<< rviz_ogre_vendor)

zzzzih, 2023-05-26:
[0.195s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'rviz_rendering' is in: /opt/ros/foxy
	'rviz_default_plugins' is in: /opt/ros/foxy
	'rviz_ogre_vendor' is in: /opt/ros/foxy
	'rviz_common' is in: /opt/ros/foxy
	'rviz2' is in: /opt/ros/foxy
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding rviz2 rviz_common rviz_default_plugins rviz_ogre_vendor rviz_rendering

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> rviz_assimp_vendor
Starting >>> rviz_ogre_vendor
Finished <<< rviz_assimp_vendor [0.39s]                                                            
[Processing: rviz_ogre_vendor]                             
[Processing: rviz_ogre_vendor]                                     
[Processing: rviz_ogre_vendor]                                       
[Processing: rviz_ogre_vendor]                                       
[Processing: rviz_ogre_vendor]                                       
[169.426s] ERROR:colcon.colcon_core.event_reactor:Exception in event handler extension 'log': [Errno 28] No space left on device [log/build_2023-05-26_18-13-26/rviz_ogre_vendor/stderr.log]
[169.426s] ERROR:colcon.colcon_core.event_reactor:Exception in event handler extension 'event_log': [Errno 28] No space left on device [log/build_2023-05-26_18-13-26/events.log]
[169.426s] ERROR:colcon.colcon_core.event_reactor:Exception in event handler extension 'log': [Errno 28] No space left on device [log/build_2023-05-26_18-13-26/rviz_ogre_vendor/stderr.log]
[169.427s] ERROR:colcon.colcon_core.event_reactor:Exception in event handler extension 'event_log': [Errno 28] No space left on device [log/build_2023-05-26_18-13-26/events.log]
[169.427s] ERROR:colcon.colcon_core.event_reactor:Exception in event handler extension 'log': [Errno 28] No space left on device [log/build_2023-05-26_18-13-26/rviz_ogre_vendor/stderr.log]
[169.601s] ERROR:colcon.colcon_core.event_reactor:Exception in event handler extension 'log': [Errno 28] No space left on device [log/build_2023-05-26_18-13-26/rviz_ogre_vendor/stdout_stderr.log]
[169.601s] ERROR:colcon.colcon_core.event_reactor:Exception in event handler extension 'log': [Errno 28] No space left on device [log/build_2023-05-26_18-13-26/rviz_ogre_vendor/stdout_stderr.log]
[169.601s] ERROR:colcon.colcon_core.event_reactor:Exception in event handler extension 'log': [Errno 28] No space left on device [log/build_2023-05-26_18-13-26/rviz_ogre_vendor/stdout_stderr.log]
[169.601s] ERROR:colcon.colcon_core.event_reactor:Exception in event handler extension 'log': [Errno 28] No space left on device [log/build_2023-05-26_18-13-26/rviz_ogre_vendor/stdout_stderr.log]
[169.608s] ERROR:colcon.colcon_core.event_reactor:Exception in event handler extension 'log': [Errno 28] No space left on device [log/build_2023-05-26_18-13-26/rviz_ogre_vendor/stdout_stderr.log]
[169.608s] ERROR:colcon.colcon_core.event_reactor:Exception in event handler extension 'log': [Errno 28] No space left on device [log/build_2023-05-26_18-13-26/rviz_ogre_vendor/stdout_stderr.log]
--- Logging error ---
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/colcon_output/event_handler/log.py", line 138, in __call__
    h.flush()
OSError: [Errno 28] No space left on device

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/colcon_core/event_reactor.py", line 78, in _notify_observers
    retval = observer(event)
  File "/usr/lib/python3/dist-packages/colcon_output/event_handler/log.py", line 143, in __call__
    raise RuntimeError(
RuntimeError: [Errno 28] No space left on device [log/build_2023-05-26_18-13-26/rviz_ogre_vendor/stdout_stderr.log]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/logging/__init__.py", line 1089, in emit
    self.flush()
  File "/usr/lib/python3.8/logging/__init__.py", line 1069, in flush
    self.stream.flush()
OSError: [Errno 28] No space left on device
Call stack:
  File "/usr/lib/python3.8/threading.py", line 890, in _bootstrap
    self._bootstrap_inner()
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3/dist-packages/colcon_core/event_reactor.py", line 68, in _run
    self._notify_observers(event)
  File "/usr/lib/python3/dist-packages/colcon_core/event_reactor.py", line 86, in _notify_observers
    logger.error(msg)
Message: "Exception in event handler extension 'log': [Errno 28] No space left on device [log/build_2023-05-26_18-13-26/rviz_ogre_vendor/stdout_stderr.log]"
Arguments: ()
[169.609s] ERROR:colcon.colcon_core.event_reactor:Exception in event handler extension 'log': [Errno 28] No space left on device [log/build_2023-05-26_18-13-26/rviz_ogre_vendor/stdout_stderr.log]
--- Logging error ---
[... 301 lines cut ...]
make[5]: *** [OgreMain/CMakeFiles/OgreMain.dir/build.make:439: OgreMain/CMakeFiles/OgreMain.dir/src/OgreCompositorInstance.cpp.o] Error 1
/home/jihyun/rviz2_ws/src/build/rviz_ogre_vendor/ogre-v1.12.10-prefix/src/ogre-v1.12.10/OgreMain/src/OgreEdgeListBuilder.cpp:566:1: fatal error: error writing to /tmp/ccdLpS3j.s: No space left on device
  566 | }
      | ^
compilation terminated.
make[5]: *** [OgreMain/CMakeFiles/OgreMain.dir/build.make:664: OgreMain/CMakeFiles/OgreMain.dir/src/OgreEdgeListBuilder.cpp.o] Error 1
/home/jihyun/rviz2_ws/src/build/rviz_ogre_vendor/ogre-v1.12.10-prefix/src/ogre-v1.12.10/OgreMain/src/OgreEntity.cpp:2506:1: fatal error: error writing to /tmp/ccJ6k4cO.s: No space left on device
 2506 | }
      | ^
compilation terminated.
make[5]: *** [OgreMain/CMakeFiles/OgreMain.dir/build.make:679: OgreMain/CMakeFiles/OgreMain.dir/src/OgreEntity.cpp.o] Error 1
make[4]: *** [CMakeFiles/Makefile2:610: OgreMain/CMakeFiles/OgreMain.dir/all] Error 2
make[3]: *** [Makefile:152: all] Error 2
make[2]: *** [CMakeFiles/ogre-v1.12.10.dir/build.make:116: ogre-v1.12.10-prefix/src/ogre-v1.12.10-stamp/ogre-v1.12.10-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/ogre-v1.12.10.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< rviz_ogre_vendor [2min 51s, exited with code 2]

Summary: 1 package finished [2min 51s]
  1 package failed: rviz_ogre_vendor
  1 package had stderr output: rviz_ogre_vendor
  6 packages not processed

how can i solve this error? i can't find this solution

Yadunund, 2023-05-26:
When opening such tickets please include more instructions on your setup and the commands you are running. Eg what OS, what ROS version is installed, whether installed from source or binaries, what packages you are trying to build, the instructions you are following etc. And please post long terminal dumps like this in a GitHub gist and share the link to the gist instead. 

Without more information it's hard to say what's happening. But one thing that is apparent from the warning at the beginning:
```
[0.195s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
'rviz_rendering' is in: /opt/ros/foxy
'rviz_default_plugins' is in: /opt/ros/foxy
'rviz_ogre_vendor' is in: /opt/ros/foxy
'rviz_common' is in: /opt/ros/foxy
'rviz2' is in: /opt/ros/foxy
```
is that you have sourced as an underlay, foxy binaries installed at /opt/ros/foxy. The workspace that you're building also contains the same packages which leads to conflicts. The build tool will compile the packages against the dependencies installed and this will fail if the version of source and installed dependencies differ.
So you should either remove the installed binaries or not source the foxy underlay when building. Either way please provide the necessary information.

cottsay, 2023-06-08:
In addition to the underlay/overlay warnings that @Yadunund discussed, I believe that the critical failure in this build is caused by your disk being full:

> No space left on device

Free up some space and try again.

clalancette, 2023-06-08:
I'm going to close this one out based on the last comment, but feel free to reopen or keep commenting if you are still having issues.

closed 2023-06-08 by clalancette