ros2/rviz #987: RViz2 can't load svg files.

xkaraman, 2023-05-30:
I was messing around with docker versions of ROS2 when i stumbled on a weird issue. The issue is that multiple warnings appear when launching RViz about loading svg files.

Using fresh docker containers:
`docker run -it --gpus all  --net host --ipc host -e DISPLAY=$DISPLAY -v $HOME/.Xauthority:/root/.Xauthority:ro ros:version`
`--ipc host` is required for Qt to display the main vis panel correctly instead of all black. (at least in my system)

==
<details><summary>ros:humble</summary>
<p>

```
apt update
apt install ros-humble-rviz2
source /opt/ros/humble/setup.bash
rviz2
```
This loads RViz default svgs correctly.
</p>
</details> 
==
<details><summary>ros:humble-perception</summary>
<p>

```
apt update
apt install ros-humble-rviz2
source /opt/ros/humble/setup.bash
rviz2
```
This failes to load RViz default svgs correctly. 
Multiple `[ERROR] [1685452528.367825331] [rviz2]: Could not load pixmap package://rviz_common/icons/move_z.svg -- using default cursor instead.` errors appear.
</p>
</details> 
==

When installing rviz2, since there is no runtime-dependency on `libqt5svg5`, it doesn't get installed and therefore, fails to load the default svgs files required.

You might think this can't be right since `humble-perception` extends `humble`. The dockerfile of [ros:humble-perception](https://github.com/osrf/docker_images/blob/20d40c96b426b8956dec203e236abff2ec29b188/ros/humble/ubuntu/jammy/perception/Dockerfile)  installs the necessary packages but doesn't install the recommended ones. `libqt5svg5` gets installed with `perception_pcl` package as a recommended one, hence it is omitted for installation and somehow when installing `rviz2` in `humble-perception` it is not recommended at all while in `humble` it is.

clalancette, 2023-06-08:
I think the difference here is likely that in the `ros:humble` case, the `apt install ros-humble-rviz2` command did the install (which implicitly install recommends), while in the `ros:humble-perception` case it was already installed by the Dockerfile (which uses --no-install-recommends).

The real fix here is that we should have an explicit dependency on the `svg` library somewhere in `rviz2` or one of its dependencies (probably whichever component is responsible for loading).

xkaraman, 2023-06-09:
> I think the difference here is likely that in the `ros:humble` case, the `apt install ros-humble-rviz2` command did the install (which implicitly install recommends), while in the `ros:humble-perception` case it was already installed by the Dockerfile (which uses --no-install-recommends).

Yeap, that is exactly the case of how it is not installed on `humble-perception` with a minor detail. RViz2 is not installed by default in `humble-perception` but some of its dependencies are and hence the `svg` library never gets installed since  it is a recommended one for `perception_pcl` package. So when instructing `apt install ros-humble-rviz2` even though we command to install both dependencies and recommended ones, since it's not a recommended package for rviz2 (how do you declare recommended in ROS packages?) it does not get installed.

> 
> The real fix here is that we should have an explicit dependency on the `svg` library somewhere in `rviz2` or one of its dependencies (probably whichever component is responsible for loading).

Adding a dependency even though is not strictly one, as it still functions correctly without it minus the correct icons, seems a good idea.

PR #992 (2023-06-09): Add a libqt5-svg dependency to rviz_common.
rviz_common is the package that loads in various icons in svg format.  In order to have them definitely be loaded in, make sure to depend on libqt5-svg.

This should fix #987
closed 2023-06-09 by ahcorde