ros2/rviz #1006: Error retrieving file

SEUZTh, 2023-07-06:
```
[rviz_rendering:debug] Available Renderers(1): OpenGL Rendering Subsystem, at /tmp/binarydeb/ros-foxy-rviz-rendering-8.2.8/src/rviz_rendering/render_system.cpp:259
[rviz_rendering:info] Stereo is NOT SUPPORTED, at /tmp/binarydeb/ros-foxy-rviz-rendering-8.2.8/src/rviz_rendering/render_system.cpp:508
[rviz_rendering:info] OpenGl version: 4.6 (GLSL 4.6), at /tmp/binarydeb/ros-foxy-rviz-rendering-8.2.8/src/rviz_rendering/render_system.cpp:242
[rviz_common:debug] Load pixmap at package://rviz_default_plugins/icons/classes/TF.svg, at /tmp/binarydeb/ros-foxy-rviz-common-8.2.8/src/rviz_common/load_resource.cpp:71
[rviz_common:debug] Error retrieving file [file:///opt/ros/foxy/share/rviz_default_plugins/icons/classes/TF.svg]: Couldn't open file /opt/ros/foxy/share/rviz_default_plugins/icons/classes/TF.svg, at /tmp/binarydeb/ros-foxy-rviz-common-8.2.8/src/rviz_common/load_resource.cpp:55
[rviz_common:debug] Load pixmap at package://rviz_default_plugins/icons/classes/TF.png, at /tmp/binarydeb/ros-foxy-rviz-common-8.2.8/src/rviz_common/load_resource.cpp:71
[rviz_common:debug] Load pixmap at package://rviz_common/icons/classes/Identity.svg, at /tmp/binarydeb/ros-foxy-rviz-common-8.2.8/src/rviz_common/load_resource.cpp:71
[rviz_common:debug] Error retrieving file [file:///opt/ros/foxy/share/rviz_common/icons/classes/Identity.svg]: Couldn't open file /opt/ros/foxy/share/rviz_common/icons/classes/Identity.svg, at /tmp/binarydeb/ros-foxy-rviz-common-8.2.8/src/rviz_common/load_resource.cpp:55
[rviz_common:debug] Load pixmap at package://rviz_common/icons/classes/Identity.png, at /tmp/binarydeb/ros-foxy-rviz-common-8.2.8/src/rviz_common/load_resource.cpp:71
[rviz_common:debug] Error retrieving file [file:///opt/ros/foxy/share/rviz_common/icons/classes/Identity.png]: Couldn't open file /opt/ros/foxy/share/rviz_common/icons/classes/Identity.png, at /tmp/binarydeb/ros-foxy-rviz-common-8.2.8/src/rviz_common/load_resource.cpp:55
[rviz_common:debug] Load pixmap at package://rviz_common/icons/default_class_icon.png, at /tmp/binarydeb/ros-foxy-rviz-common-8.2.8/src/rviz_common/load_resource.cpp:71
```
After I found some pictures and placed them in the corresponding folder, the QT interface still cannot be displayed.

```
[rviz_rendering:debug] Available Renderers(1): OpenGL Rendering Subsystem, at /tmp/binarydeb/ros-foxy-rviz-rendering-8.2.8/src/rviz_rendering/render_system.cpp:259
[rviz_rendering:info] Stereo is NOT SUPPORTED, at /tmp/binarydeb/ros-foxy-rviz-rendering-8.2.8/src/rviz_rendering/render_system.cpp:508
[rviz_rendering:info] OpenGl version: 4.6 (GLSL 4.6), at /tmp/binarydeb/ros-foxy-rviz-rendering-8.2.8/src/rviz_rendering/render_system.cpp:242
[rviz_common:debug] Load pixmap at package://rviz_default_plugins/icons/classes/TF.svg, at /tmp/binarydeb/ros-foxy-rviz-common-8.2.8/src/rviz_common/load_resource.cpp:71
[rviz_common:debug] Load pixmap at package://rviz_common/icons/classes/Identity.svg, at /tmp/binarydeb/ros-foxy-rviz-common-8.2.8/src/rviz_common/load_resource.cpp:71
```

clalancette, 2023-07-13:
A couple of things that will help us help you here:

First, Foxy is End-of-Life as of last month.  If you could try this with a newer distribution (like Humble or Iron), that would be helpful.

Second, we need more information about what you are trying to do.  We'll need what ROS distribution you are using, your system information, and package information.  Then we also need some idea of the packages you are trying to build/use, and how you are accomplishing that.  With that information, we may be able to provide more help.

1192499095, 2024-12-17:
I'm using ubuntu22.04,ros2_humble and tried to integrate rviz2 with my program into qt5.15 and had the same problem.

mjlee111, 2025-03-21:
> I'm using ubuntu22.04,ros2_humble and tried to integrate rviz2 with my program into qt5.15 and had the same problem.
Did you managed to fix the problem?