ros2/rviz #1024: border padding around main visualization window, not going away in fullscreen

altineller, 2023-07-27:
Hello,

How can I make the border around main visualzation window go away? Or change its color if we cant make it go away.
When in full screen it apppears as a 8-9px white border.

I understand that there is no setting or customization for this, and I already have built rviz from source to get the default full screen launch going, and now I would like to remove the border or at least set it color to black. 

Which part of code is responsible for this? I have been reading the source code, and could it be the border of a remote library or object?

Any ideas/help greatly appreciated.

Best Regards,
C.A.

clalancette, 2023-08-03:
Unfortunately, we don't have a great idea on how to do this.  Likely this has something to do with the Qt settings or the theme, but I'm not sure.  If you do figure something out, please feel free to update here or open a PR to improve things.

altineller, 2023-08-06:
@clalancette  I think it has to do with te OgreRenderWindow.h, which is at ./ros/humble/opt/rviz_ogre_vendor/include/OGRE/OgreRenderWindow.h

bochen87, 2024-07-09:
> @clalancette I think it has to do with te OgreRenderWindow.h, which is at ./ros/humble/opt/rviz_ogre_vendor/include/OGRE/OgreRenderWindow.h

How did you solve this issue?

PR #1228 (2024-07-22): Set ContentsMargins for RenderPanel to 0 to avoid borders in fullscreen mode
Fixes #1024
commit 642b1a34 (2024-07-22): Set ContentsMargins for RenderPanel to 0 to avoid borders in fullscreen mode. Fixes #1024 (#1228)
PR #1250 (2024-07-24): Set ContentsMargins for RenderPanel to 0 to avoid borders in fullscreen mode (backport #1228)
Fixes #1024<hr>This is an automatic backport of pull request #1228 done by [Mergify](https://mergify.com).
commit b158842c (2024-07-24): Set ContentsMargins for RenderPanel to 0 to avoid borders in fullscreen mode. Fixes #1024 (#1228) (#1250)

(cherry picked from commit 642b1a34b1a50bfa8afa76817ec3fe0e2f9cf210)
closed 2024-07-22 by ahcorde