ros2/rviz #1025: Old config file causes crash in Iron

dcconner, 2023-08-01:
From https://answers.ros.org/question/417874/rviz2-crashing-in-iron/

My old file from Humble caused crash when I switched to Iron.  Deleting the file allowed me to load RViz.

I suggest trying to catch any issues on configuration load and provide suitable error message to user.

Unfortunately, I don't have time to work on this so I suspect this is a future enhancement.

christophfroehlich, 2023-08-17:
I have the same problem, and narrowed it down to `rviz_default_plugins/Camera` in the rviz file causing the segmentation fault. Even if I remove this, add the camera plugin again, save and reopen causes the same crash.

```yaml
    - Class: rviz_default_plugins/Camera
      Enabled: false
      Far Plane Distance: 100
      Image Rendering: background and overlay
      Name: Zed Camera
      Overlay Alpha: 0.5
      Topic:
        Depth: 5
        Durability Policy: Volatile
        History Policy: Keep Last
        Reliability Policy: Best Effort
        Value: /zed2i/left/image_rect
      Value: false
      Visibility:
        Collision Bodies: true
        Grid: true
        Log Detections: true
        Log Ground Truth: true
        MarkerArray: true
        Path: true
        Picked Log: true
        RobotModel: true
        TF: true
        Target: true
        True Path: true
        Value: true
      Zoom Factor: 1
```

dghw, 2023-08-29:
Similar issue here - but I found this thread when searching for bugs with the Camera plugin in iron/rviz2. When I start a new rviz instance and add a Camera plugin it's initially fine until I select the topic (/image_raw). As soon as I select this topic rviz crashes. 

Exact same setup works when starting rviz2 from a terminal with Humble sourced instead of Iron, & the camera image can be visualised just fine.