ros2/rviz #1052: rviz2 screen flickering when enable HiDPI

byeongkyu, 2023-08-30:
ROS2 distro: Humble
Issue: rviz2 screen flickering when enable HiDPI

When running rviz2, if environment variables such as QT_ENABLE_HIGHDPI_SCALING that activate hidpi are set, the main 3D screen flickers severely. Of course, one can use the option QT_ENABLE_HIGHDPI_SCALING=0 to deactivate hidpi, but then the proportions of the internal components are slightly off. I'm curious about what's causing this issue.

clalancette, 2023-09-07:
We don't have a great idea on what could be causing this.  It could be a graphics driver problem or maybe something in one of the underlying libraries (OGRE or Qt).  Can you please give us a lot more information about your platform, including the graphics card, driver, Linux kernel, etc?

byeongkyu, 2023-09-10:
@clalancette In my opinion, this doesn't seem to be a problem unique to me. It's a common issue on all HiDPI systems I've experienced. The graphics card I'm currently using is Nvidia, and the issue occurs on both the 20 series and 30 series. I will attach detailed spec information.

OS: Ubuntu 22.04.3 LTS x86_64
CPU: 12th Gen Intel i9-12900KF (24) @ 5.100GHz
GPU: NVIDIA 01:00.0 NVIDIA Corporation Device 2203
DE: GNOME 42.9 
Shell: bash 5.1.16
Kernel: 6.2.0-26-generic

fprott, 2023-10-20:
> When running rviz2, if environment variables such as QT_ENABLE_HIGHDPI_SCALING that activate hidpi are set, the main 3D screen flickers severely. 

I also got that flickering issue but changing the QT_ENABLE_HIGHDPI_SCALING variable seems not to help. Can you explain how you set the system variable? Just `export QT_ENABLE_HIGHDPI_SCALING=0` has no effect for me.

byeongkyu, 2023-10-23:
@fprott 

When using a high-resolution monitor (4K), it's common to set the display scale to 200%. If the QT_ENABLE_HIGHDPI_SCALING option is not used in this case, the size of Qt's resources may not match, leading to misaligned UI sizes. Of course, it's possible to use it without turning on the QT_ENABLE_HIGHDPI_SCALING option, but isn't it necessary for a more beautiful UI?

vembacher, 2023-11-10:
> I also got that flickering issue but changing the QT_ENABLE_HIGHDPI_SCALING variable seems not to help. Can you explain how you set the system variable? Just `export QT_ENABLE_HIGHDPI_SCALING=0` has [...]

Changing `QT_SCREEN_SCALE_FACTORS` to `1` for each display output worked for me when `QT_ENABLE_HIGHDPI_SCALING` did not help.

Just `echo` the variable, then export it with the altered value and finally launch `rviz2` from the same shell.

TheRealBeef, 2024-05-13:
Cross-posting from the related topic.

I have the same issue, but it persists regardless of the DPI setting for me. I also tried setting via commandline the QT_SCREEN_SCALE_FACTORS=1 and also unsetting the scale factors as well.

I'm running Ubuntu 22.04 on a laptop with Nvidia GPU and this happens both in Mate and in KDE desktops - although with KDE I can do what's in the original post of issue #1079 and stop compositing so it doesn't flicker - so long as I don't alter the size of the windows. Once they've been altered in any way then it starts flickering and won't stop.

It flickers not just the world visualization but also any images that are displayed in their own boxes as well.

themightyoarfish, 2024-09-30:
For me, unsetting `QT_SCALE_FACTOR` (was 2) makes it not flicker, but then the UI is obviously too small to really interact with.

christian-rauch, 2026-08-19:
I encountered the same issue on Ubuntu 26.04. I did not see this on Ubuntu 24.04.

The solution in https://github.com/ros2/rviz/issues/1079#issuecomment-4739501556 works for me:
```
QT_ENABLE_HIGHDPI_SCALING=0 rviz2
```

Only the icons are a bit to small. The text is scaled correctly.

vahapt, 2026-09-20:
Confirmed #1870 resolves the issue. 

| Component | Detail |
| --- | --- |
| **OS / Distro** | Ubuntu 26.04.1 LTS (Resolute Raccoon) aarch64 |
| **Kernel** | `7.3.0-070300rc2-generic` - patched|
| **Desktop ** | GNOME Shell 50.1 (Wayland) — rviz2 on XWayland |
| **Device Model** | ASUS Zenbook A16 (UX3607OA) |
| **SoC** | Qualcomm Snapdragon X2 Elite, 18 cores |
| **Graphics** | Adreno X2-85 — freedreno / Mesa `26.0.8-1ubuntu0.3` |
| **Memory** | 48 GB |
| **Display** | eDP-1 2880x1800 @ 120 Hz, **150% fractional scale** |
| **Qt** | 5.15.18 |
| **ROS** | ROS `lyrical`|