ros2/rviz #1061: How to set the Z coordinate information in the initial pose?

goruchanchan, 2023-09-09:
## What I want

When using the 2D Pose Estimate tool to input 2D initial pose(x,y,θ), I would like to be able to adjust the height(z) synchronously as well.

## Why need?

In the case of 3D pose estimation, the height must also be adjusted to set the initial pose properly.

The current method is achieved by flying geometry_msgs/msg/PoseWithCovarianceStamped type data by ros2 topic pub by itself.

Specifically, it is as follows.

1. Check the value adjusted by Rviz2 for 2D pose.
2. Adjust that data to /initialpose.
3. Adjust the height (z) of [2. data].
4. Send [3. data] by ros2 topic pub.

The workflow of steps 2 and 3 is complicated, so I am wondering if RViz2 can provide height to improve the workflow.

Or if there is an efficient way to do this, I would like to know.

## Actual Result

Currently, it is fixed at 0.

[image]

clalancette, 2023-09-21:
The 2D pose estimation tool is exactly what it says: for 2 dimensions.

There could in theory be a 3D pose estimation tool, but it isn't clear how you would design the user interaction for this.  You can't just do it based on a click on the canvas, since that is only in 2 dimensions.  One idea to do this is to have a 2D point, but then also have a slider or something like that for adjusting the Z coordinate (the drone community does something like this).

Another way to do this in a kind of hacky way is to introduce another node that listens to the (remapped) `/initial_pose`, adds in a Z-coordinate, and then republishes onto `/initial_pose`.

We aren't likely to look into a new 3D pose estimation tool in the near future, so I'm putting this on the backlog.  We'd be happy to review pull requests that added it, though.

goruchanchan, 2023-09-22:
Thanks for the response.

> it isn't clear how you would design the user interaction for this.

You are right, that is exactly what I was thinking. The example of the drone community was very helpful! Thanks! I will take a look at that one as well!

> We'd be happy to review pull requests that added it, though.

We'd love to have the opportunity to do so again. Thank you very much for this time!