ros2/rviz #1066: Cannot open rviz2

TannnYannng, 2023-09-27:
$ rviz2
[INFO] [1695823660.091830699] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1695823660.091943524] [rviz2]: OpenGl version: 4.1 (GLSL 4.1)
D3D12: Removing Device.
Segmentation fault

Could someone help?

clalancette, 2023-09-27:
Can you please give details on what ROS release you are using, and which platform you are using?

TannnYannng, 2023-09-28:
> Can you please give details on what ROS release you are using, and which platform you are using?

Hi, I am using ros2 humble and wsl2

clalancette, 2023-09-28:
> Hi, I am using ros2 humble and wsl2

OK, yeah, I thought it would be WSL2.  That's not currently one of our supported platforms, so I don't have a lot of advice to give you.  It will probably be best to open a question on https://robotics.stackexchange.com/questions/tagged/ros, which is our central Question and Answer site.

If you find out a way to fix it, we'd be happy to take pull requests to update the code or the documentation to make it run.

TannnYannng, 2023-09-28:
> OK, yeah, I thought it would be WSL2. That's not currently one of our supported platforms, so I don't have a lot of advice to give you. It will probably be best to open a question on [...]

OK thanks^^ I will do it

TannnYannng, 2023-10-10:
fixed the issue by adding 
`export LIBGL_ALWAYS_SOFTWARE=true`
to .bashrc file.
Thanks! @clalancette

closed 2023-10-10 by TannnYannng

clalancette, 2023-10-10:
Thanks!  I've made a PR to the documentation to add this as a possible troubleshooting step: https://github.com/ros2/ros2_documentation/pull/3965