ros2/rviz #1067: Saved map not loading in Rviz2

Jyoshrobot1234, 2023-10-03:
Hi everyone,
ROS- Humble
Ubuntu-22.04

I am trying to load a saved map using slam toolbox and i am getting the following error in rviz2 and at the same time a new map starts generating
[image]

Thanks for helping

Jyoshrobot1234, 2023-10-03:
@Martin-Idel  can you please help.

sloretz, 2023-10-12:
Would you be willing to post a minimal reproducible example? That will help the maintainers understand what's crashing and why.

mikeferguson, 2023-11-24:
Since it is a screen shot, this doesn't show up in a search for "rviz/glsl120/indexed_8bit_image.vert", but is possibly related/similar to https://github.com/ros2/rviz/issues/463

koa-ellie, 2024-12-08:
@Jyoshrobot1234 did you solve your problem because I'm facing this problem right now