ros2/rviz #1077: Performance issues with large PointClouds

FabianThomsen, 2023-10-12:
I have a PointCloud2 message with 6716596 points, colored in RGB. Publishing it only once and displaying it in RVIZ2 already leads to massive performance drops. When changing the Style e.g. to Boxes, it becomes totally impossible to move the view around.

Due to the large number of points, I would say, this may be expected - but in RVIZ in ROS1, the same works without any lags on my machine, so I guess there is some substantial difference between RVIZ and RVIZ2 leading to this issue.

sloretz, 2023-11-02:
Would you be willing to provide a super small bag with that point cloud for testing purposes?

FabianThomsen, 2023-11-03:
> Would you be willing to provide a super small bag with that point cloud for testing purposes?

Sure, you can download a bag containing the pointcloud twice (in an interval of 100s) and rviz config from [https://gigamove.rwth-aachen.de/de/download/327c66f9fc885f4d74c46eb5cde8eee2](here) within the next two weeks. The bag was recorded in humble.

savalena, 2025-05-13:
I face the same problem as @FabianThomsen. 

My PointCloud is static and has 6536791 uncolored points. When I visualize it in rviz, the interaction in rviz gets quite laggy. Changing the style of points to any none-default style leads to more lags, I barely can interact in rviz. In ROS1 I used the same PointClouds and there were no performance issues. Does anyone have any suggestions how can this problem be addressed?

shrijitsingh99, 2025-05-24:
I too see a similar issue

OxygenLost, 2026-01-05:
I too see a similar issue