ros2/rviz #1078: rviz2 segfault with error 4 in libc.so.6[7f8b852c4000+195000

ionutnechita-intel, 2023-10-18:
Hi ROS Team,

I notice today one issue regarding RVIZ.
ROS2 RVIZ is terminated with segfault on Ubuntu 22.04 LTS and Humble ROS distro.

This is not happened sporadically, occurs quite often.

[  848.926347] rviz2[6944]: segfault at 7f894c943040 ip 00007f8b8543c9cd sp 00007ffe616babc8 error 4 in libc.so.6[7f8b852c4000+195000]
[  848.926356] Code: 00 00 00 00 00 66 66 2e 0f 1f 84 00 00 00 00 00 66 66 2e 0f 1f 84 00 00 00 00 00 66 90 f3 0f 1e fa 48 89 f8 48 83 fa 20 72 23 <c5> fe 6f 06 48 83 fa 40 0f 87 a5 00 00 00 c5 fe 6f 4c 16 e0 c5 fe

```
dpkg -l | grep rviz
ii  ros-humble-nav2-rviz-plugins                      1.1.12-1jammy.20231005.220631                         amd64        Navigation 2 plugins for rviz
ii  ros-humble-rtabmap-rviz-plugins                   0.21.1-1jammy.20230920.004632                         amd64        RTAB-Map's rviz plugins.
ii  ros-humble-rviz-assimp-vendor                     11.2.8-1jammy.20230919.195205                         amd64        Wrapper around assimp, providing nothing but a dependency on assimp, on some systems.
ii  ros-humble-rviz-common                            11.2.8-1jammy.20230919.225604                         amd64        Common rviz API, used by rviz plugins and applications.
ii  ros-humble-rviz-default-plugins                   11.2.8-1jammy.20230919.232358                         amd64        Several default plugins for rviz to cover the basic functionality.
ii  ros-humble-rviz-ogre-vendor                       11.2.8-1jammy.20230919.195220                         amd64        Wrapper around ogre3d, it provides a fixed CMake module and an ExternalProject build of ogre.
ii  ros-humble-rviz-rendering                         11.2.8-1jammy.20230919.203743                         amd64        Library which provides the 3D rendering functionality in rviz.
ii  ros-humble-rviz2                                  11.2.8-1jammy.20230920.004636                         amd64        3D visualization tool for ROS.

```

sloretz, 2023-11-02:
Thanks for the report! Would you be willing to provide a reproducible example? What is RViz visualizing when the crash happens, and what inputs (if any are being given to it?) Otherwise if you're willing to build from source a stack trace would be very helpful in resolving this.

ionutnechita-intel, 2023-11-07:
Using gdb i have this output:

```
                Found module /opt/ros/humble/bin/rviz2 with build-id: d6e70b77f8651db6ba24f34dd4ab64e6accd3e34
                Stack trace of thread 10796:
                #0  0x00007fed327aefcd n/a (n/a + 0x0)

GNU gdb (Ubuntu 12.1-0ubuntu1~22.04) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /opt/ros/humble/bin/rviz2...
Reading symbols from /usr/lib/debug/.build-id/d6/e70b77f8651db6ba24f34dd4ab64e6accd3e34.debug...
BFD: warning: /var/lib/systemd/coredump/core.rviz2.1000.9106313dced046d4a5d17e231221e776.10796.1699373855000000 has a segment extending past end of file

warning: Can't open file /dev/shm/fastrtps_8b88d13b992ec466 (deleted) during file-backed mapping note processing
[New LWP 10796]
[New LWP 10798]
[New LWP 10799]
[New LWP 10800]
[New LWP 10801]
[New LWP 10802]
[New LWP 10805]
[New LWP 10813]
[New LWP 10807]
[New LWP 10810]
[New LWP 10815]
[New LWP 10806]
[New LWP 10808]
[New LWP 10809]
[New LWP 10816]
[New LWP 10803]
[New LWP 10814]
[New LWP 10812]
[New LWP 10817]
[New LWP 10811]
[New LWP 10819]
[New LWP 10823]
[New LWP 10820]
[New LWP 10822]
[New LWP 10804]
[New LWP 10825]
[New LWP 10818]
[New LWP 10821]
[New LWP 10824]

warning: Error reading shared library list entry at 0x29ba00746473
Failed to read a valid object file image from memory.
Core was generated by `rviz2 -d realsense_config.rviz'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007fed327aefcd in ?? ()
[Current thread is 1 (LWP 10796)]
```