ros2/rviz #1079: rviz2 major screen flickering/tearing

fprott, 2023-10-20:
Hello,

As the following video shows I experience a critical bug with rviz2. The screen is flickering/tearing even when not displaying any data. This will happen even if I don't resize!

Basically as soon as I start rviz2 I get the following flickering only inside the 3D scene (not the surrounding tools!).

https://github.com/ros2/rviz/assets/24491505/f4c81b15-5d3f-4149-a52a-ae4f48401249

This is NOT a monitor issue, this recording was not filmed off the monitor but I used a screen-grabbing software.

Now it appears to be a frame rate issue. When I play around with the frame rate, then save that option and then restart rviz the bug will no longer appear till I resize or switch out of the window in any way. I suspect that on the time of resizing rviz forgets the frame rate?

So I suspect the problem is that Compositor is at fault. It turns out that if I unset the option "Allow applications to block compositing", as seen in the attached screenshot, the problem will no longer appear till the window is resized. So Compositor can't properly work?

Anyway, with this remedy I can make RVIZ2 barely usable but it is not really a good experience. Without it this is totally system breaking and might cause epilepsy or something.

My specs:
cpu: i9-13900HX
monitor: ASUS VG34V
gpu: NVIDIA GeForce RTX 4080 Laptop GPU/PCIe/SSE2
VGA compatible controller: NVIDIA Corporation Device 27e0 (rev a1)
NVRM version: NVIDIA UNIX x86_64 Kernel Module  530.41.03  Thu Mar 16 19:48:20 UTC 2023

sloretz, 2023-11-02:
Does this happen with any other OpenGL windows? It seems like something that might be a bug with wayland or X11 or the QT version on the system.

mw46d, 2024-02-01:
Same problem:-(
CPU: i9-13980HX
GPU: NVIDIA GPU NVIDIA RTX 2000 Ada Generation Laptop GPU 
Kubuntu 22.04 & Humble fully updated (running X-server)
Nvidia driver: 545.29.06

glmark2 (defaults to Mesa/Intel GPU) works fine
__NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia glmark2 (to force the nvidia GLX/GL implementation) works as well 

rviz2 and
__NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia rviz2
both show the flickering:-(  

With LD_DEBUG=files ..., you can actually see, which shared libraries are loaded 'during startup' of the process;-)

DerProfessor, 2024-02-23:
For me it works to set `QT_SCREEN_SCALE_FACTORS=1` before running `rviz2`. This seems to be related to #1052.

mw46d, 2024-02-23:
Yeah;-) or setting it to '1' archives the same thing. No scaling for HighRes monitors. I created a little rviz2 script, so I don't forget;-) 
```marcow@feather7:~$ cat bin/rviz2 
#!/usr/bin/bash

unset QT_SCREEN_SCALE_FACTORS
exec /opt/ros/$ROS_DISTRO/bin/rviz2 "$@"
```

macmacal, 2024-02-29:
I had similar flickering problems on Kubuntu 22.04 on laptop Lenovo Legion Pro7 (Ryzen 9 7945HX RTX 4080).

After disabling the integrated AMD graphics I run into a problem with the global scale on Qt applications (which seems related to [this issue](https://forum.manjaro.org/t/kde-plasma-100-global-scale-does-not-work-for-qt-apps/139347)).  To overcome this I just set the global scale to the minimal value above 100% (i.e. 106,25%):

[image]

After that modifications RViz2 was still working fine, until some system update for Plasma / ROS Humble packages.

> For me it works to set `QT_SCREEN_SCALE_FACTORS=1` before running `rviz2`. This seems to be related to #1052.

This works like a charm, many thanks!

TheRealBeef, 2024-05-13:
Interestingly, I have the same issue, but it persists regardless of the DPI setting for me. I also tried setting via commandline the `QT_SCREEN_SCALE_FACTORS=1` and also unsetting the scale factors as well.

I'm running Ubuntu 22.04 on a laptop with Nvidia GPU and this happens both in Mate and in KDE desktops although with KDE I can do what's in the original post and stop compositing so it doesn't flicker so long as I don't alter the size of the windows (not sure yet how to do that in Mate).

It flickers not just the world visualization but also any images that are displayed in their own boxes as well.

Frank1126lin, 2024-06-03:
Same issue, adjust the display setting screen scale to 1, that problem fixed. But the words get too small to see clearly.

themightyoarfish, 2024-09-30:
Same issue with an ubuntu mate install, QT_SCREEN_SCALE_FACTORS=1 did not do anything.

mw46d, 2025-12-04:
My current little script is

```
marcow@feather7:~$ cat bin/rviz2 
#!/usr/bin/bash

export ROS_DISTRO=${ROS_DISTRO:-jazzy}
export QT_SCREEN_SCALE_FACTORS=1
export QT_ENABLE_HIGHDPI_SCALING=0
export __NV_PRIME_RENDER_OFFLOAD=1
export __GLX_VENDOR_LIBRARY_NAME=nvidia

exec /opt/ros/$ROS_DISTRO/bin/rviz2 "$@"
```

But don't forget to also change the two places in **/opt/ros/*** !! ;-) Otherwise `ros2 run ...rviz2 ...` might not use the PATH, you expect! Unfortunately, package updates will override those changes.

```
marcow@feather7:~$ ls -l /opt/ros/jazzy/bin/rviz2*
-rwxr-xr-x 1 root   root 27144 Nov  7 16:48 /opt/ros/jazzy/bin/rviz2
-rwxr-xr-x 1 marcow uucp   328 Nov 18 21:54 /opt/ros/jazzy/bin/rviz2.mw
-rwxr-xr-x 1 root   root 27144 Oct  7 17:21 /opt/ros/jazzy/bin/rviz2.orig
marcow@feather7:~$ ls -l /opt/ros/jazzy/lib/rviz2/rviz*
-rwxr-xr-x 1 root   root 16895 Nov  7 00:56 /opt/ros/jazzy/lib/rviz2/rviz1_to_rviz2.py
-rwxr-xr-x 1 root   root 27144 Nov  7 16:48 /opt/ros/jazzy/lib/rviz2/rviz2
-rwxr-xr-x 1 marcow uucp   334 Nov 18 21:56 /opt/ros/jazzy/lib/rviz2/rviz2.mw
-rwxr-xr-x 1 root   root 27144 Oct  7 17:21 /opt/ros/jazzy/lib/rviz2/rviz2.orig
```

I hope, that helps;)

nkminion, 2025-12-29:
```bash
alias rviz2="QT_SCREEN_SCALE_FACTORS=1 QT_FONT_DPI=144 rviz2"
```

Using ```QT_SCREEN_SCALE_FACTORS=1``` made it hard for me to see anything so I used ```QT_FONT_DPI=144```

christian-rauch, 2026-08-20:
The issue seems related to scaling in Xwayland.

I can also prevent flickering in GNOME by disabling the "Built-in Scaling for Legacy Apps":

[image]

or via command line:
```sh
gsettings set org.gnome.mutter.wayland xwayland-scaling-factor 1
```

See also https://www.omgubuntu.co.uk/2026/03/gnome-50-released.

mw46d, 2026-08-20:
I don't use Wayland with my Kubuntu 24.04 [yet]. I believe, the issue is somewhere on the X11 + driver side.