ros2/rviz #1082: RVIZ2 doesn't work on Windows afer fresh install: missing DLL `zlib1.dll`

RobertoRoos, 2023-11-02:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10 / Sandbox Windows 10
- Installation type:
  - binaries
- Version or commit hash:
  - `ros2-humble-20230724-windows-release-amd64`
- DDS implementation:
  - default
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```
ros2 run rviz2 rviz2
```

#### Expected behavior

RVIZ2 will start.

#### Actual behavior

RVIZ2 does not start, there is only the message:

> [ros2run]: Process exited with failure 3221225595

`rviz2.exe` just exists with a negative error code.

#### Additional information

On my fresh install, RVIZ2 cannot run. It exits with a negative exit code without further explanation. More information and the workaround in my stackexchange question: https://robotics.stackexchange.com/questions/104892/ros2-humble-on-windows-10-rviz2-gives-error-process-exited-with-failure-32212

It turns out it's looking for a DLL named `zlib1.dll`, and Windows does not resolve it to `zlib.dll`.

Apologies if this is not the correct issues list. Since this seems like an error in the publication step, I figured it'd fit here best.

sloretz, 2023-11-16:
@RobertoRoos would you be willing to attach the contents of C:\xmllint\bin? There was some discussion that the issue you're seeing might be a conflict between chocolatey packages and these instructions: https://docs.ros.org/en/humble/Installation/Alternatives/Windows-Development-Setup.html#install-miscellaneous-prerequisites

RobertoRoos, 2023-11-17:
Sure, here: 
[xmllint_bin.zip](https://github.com/ros2/rviz/files/13389432/xmllint_bin.zip)

However, I'm pretty sure I only installed xmllint _after_ I had this issue with RVIZ2. Only after this I also tried to build ROS2 from source (for different reasons).
xmllint is not listed under the "Windows (binary)" instructions. But I see that `xmllint/bin` indeed contains a `zlib1.dll`, which I guess would have also solved this issue.

seanfuture, 2023-12-07:
Also faced the same issue / error as above.  First problem is [Windows (binary) installation instructions](https://docs.ros.org/en/humble/Installation/Windows-Install-Binary.html) doesn't make any reference to xmllint or rviz, which I would think would be quite instrumental.  After extracting the dependencies into `C:\xmllint\bin` the cannot find `zlib1.dll` dialog box went away, but still currently getting the following error when running `ros2 run rviz2 rviz2`:

`[ros2run]: Process exited with failure 3221225781`

dirksavage88, 2024-02-14:
I am able to replicate the same issues with zlib1.dll not being found. I even tried updating/restoring Visual Studio. Ended up downloading the dll into the system folders but also got the windows error. I am able to get foxglove to work reasonably well in windows for the time being, but this should be fixed.

zhaolj214, 2024-04-02:
1. using chocolatey install ros2-humble-desktop.221202-pre.nupkg
 2. enter the default installation path,  /opt/ros/humble/x64/bin
 3. zlib1.dll can be found here

dirksavage88, 2024-05-25:
Update: I just installed ros2 jazzy on windows 10 and was able to get rviz2 to come up with no issues