ros2/rviz #1108: Possible to vizualize a mesh not in the filesystem of the machine running RVIZ ?

robinpassama, 2023-12-15:
Hi ROS team and sorry if this problem has already been raised (did not find any thread about that) !

**Problem:** I cannot visualize mesh from visuals used in robot_description when I run rviz from another PC that the one "hosting the URDF". 

**Context:** I currently use the `robot_description` **topic** (not the parameter) to provide system description to rviz. This is mandatory in my context since URDF is dynamically generated. The URDF description contains path to mesh files used for visuals and collisions. Everything work well as long as I run RVIZ on the machine hosting the node that generates the URDF. But when I run it in another machine  mesh files are not installed in this machine's filesystem so they cannot be shown by RVIZ.

**Invalid Solution:** I know that if I "clone" the ROS workspace on the remote machine to get meshes installed, then of course everything will work (if I used package relative path), but this is not a valid situation in my context. First, IMO generally speaking the remote machine is not supposed to have everything installed, Second, and more problematic, URDFs are dynamically generated and contain reference to mesh files unknown on remote machine when rviz is launched. 

**Need:** The simplest solution would be to have a way to transfer data files to RVIZ any time the robot description topic is updated.

**Question:** Is there any way to automatically make rviz load data files referenced in the `robot_description` topic ? Or another way to provide/update a complete robot description to rviz at runtime ?

Thanks for reading !

clalancette, 2023-12-20:
> **Question:** Is there any way to automatically make rviz load data files referenced in the `robot_description` topic ? Or another way to provide/update a complete robot description to rviz at runtime ?

Unfortunately, not that I know of.  I guess this would be an interesting feature to have, so if you are interested in trying to make it work, that would be great.  If you do decide to do this, I'll suggest doing a quick design and posting it here so we can iterate over it before you start implementing anything.

robinpassama, 2024-01-08:
I do not have much time to do that but I can try.

**Design Principle:**

1) keep the robot description unchanged, only the value of the `mesh` tag's `filename` argument is modified to target a non local mesh. The idea is to add a new type of path, that reference a topic, something like:

```
<mesh filename="topic:/the/mesh/message">
```
IMO the advantage of this solution is that the XML parsing is left unchanged. Only the parsing of the `filename` argument must be customized.

2) When a code publish a `robot_description` it should also publish topics used in this description. Topics are published in "transient local" equivalent QoS mode. Each topic has `M̀esh` interface following convention described here: https://github.com/ros2/common_interfaces/blob/humble/shape_msgs/msg/Mesh.msg.

3) When RVIZ reads the `robot_description`, it parses the `filename` argument of mesh and then if the path given starts with a `topic` keyword it loads the mesh by reading the topic instead of reading a file.

**Question:**
- is there any ROS2 library providing functionalities for loading a mesh in memory (from a file) and for serializing/deserializing meshes into/from `Mesh` data type ?

Do you think this approach makes sense ?