ros2/rviz #1110: Error when adding display type more than 3

evertale888, 2023-12-25:
Run the latest humble rviz2, when adding more than 2 display type (add occupancy grid, occupancy map & map) , i got this error below..

[ERROR] [1703473400.309093725] [rviz2]: ItemIdentityException: Texture with the name MapPaletteTexture0 already exists. in ResourceManager::add at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreResourceManager.cpp (line 148)

Any help will be appreciated.

clalancette, 2024-01-04:
@ahcorde when you get a chance, can you take a look here and see if you can reproduce the problem?

ahcorde, 2024-01-05:
Hi @evertale888, 

I was trying to reproduce your issue but I'm not able.

Can you share a gif o video about how to reproduce this failure ?

evertale888, 2024-01-05:
> Hi @evertale888,
> 
> I was trying to reproduce your issue but I'm not able.
> 
> Can you share a gif o video about how to reproduce this failure ?

1st, i run it on ubuntu 22.04 with ros2 humble. Im not build it from source, just use the one come with humble. So, what i did was,  I just run emplty/plain rviz with usual command 'ros2 run rviz2 rviz2. Then, i try to add display type occupancy grid, occupancy map & map. When i try to add the 3rd one, it cause the error as mention above. Even if i try different adding order, the 3rd one will still cause that error. Even if i use rviz2 that connected any ros2 node, for example the octomap server, its still resulted in the same error when i try add more than 2 display type. So to reproduce just run plain rviz2 normally & try add those display type.

2805651606, 2024-07-12:
> > 你好@evertale888，
> > 我试图重现您的问题但未能成功。
> > 您能分享一个关于如何重现此故障的 gif 或视频吗？
> 
> 首先，我在 ubuntu 22.04 上使用 ros2 humbiz 运行它。我不是从源代码构建它，只使用 humbiz 自带的那个。所以，我所做的是，我只是用通常的命令“ros2 run rviz2 rviz2”运行空/普通的 rviz。然后，我尝试添加显示类型占用网格、占用图和地图。当我尝试添加第三个时，它会导致如上所述的错误。即使我尝试不同的添加顺序，第三个仍然会导致该错误。即使我使用连接任何 ros2 节点的 rviz2，例如 octomap 服务器，当我尝试添加超过 2 种显示类型时，它仍然会导致相同的错误。因此，要重现，只需正常运行普通的 rviz2 并尝试添加这些显示类型。

你好，请问你解决这个问题了？我现在遇到的情况跟你一样。我的也是ubuntu22,ros2-hubmle

2805651606, 2024-07-12:
> > 你好@evertale888，
> > 我试图重现您的问题，但未能成功。
> > 您能分享一个关于如何重现此故障的 gif 或视频吗？
> 
> 首先，我在 ubuntu 22.04 上使用 ros2 humbiz 运行它。我就是从源代码构建它，只使用 humbiz 自带的那个。所以，我说的是，我只是用通常的命令“ros2 run rviz2 rviz2”运行空/普通的 rviz。然后，我尝试添加显示类型占用网格、占用图和地图。当我尝试添加第三个时，会导致如上所述的错误。即使我尝试不同的添加顺序，第三个仍然会导致该错误。即使我使用连接任何 ros2 节点的 rviz2，例如 octomap 服务器，当我尝试添加超过 2 种显示类型时，它仍然会导致相同的错误。因此，要重现，只需尝试添加这些普通的 rviz2 并尝试添加这些显示类型。

你好，请问你解决了？我现在遇到条件跟你一样。我的也是ubuntu22，ros2-hubmle

evertale888, 2024-07-29:
> 你好，请问你解决了？我现在遇到条件跟你一样。我的也是ubuntu22，ros2-hubmle

Hello there, unfortunately i can't manage to solve it.. i use ros1 for now.

gyyc233, 2026-01-21:
I suffer the same situation