ros2/rviz #1111: QStandardPaths: wrong permissions on runtime directory /run/user/1000/ 0755 instead of 0700

jiaweiz9, 2023-12-26:
When running rviz2 on WSL2:
QStandardPaths: wrong permissions on runtime directory /run/user/1000/, 0755 instead of 0700
[INFO] [1703573194.517531901] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1703573194.517620901] [rviz2]: OpenGl version: 4.2 (GLSL 4.2)
[INFO] [1703573194.556220302] [rviz2]: Stereo is NOT SUPPORTED

Same issue happens when running rqt, however running gazebo-classic is okay.

platform: windows 11 wsl2 Ubuntu 22.04.3 LTS (Jammy Jellyfish)
ros2 version: humble

It looks like something wrong when running Qt libraries. Qt needs more permissions on that directory? Really appreciate any suggestions

clalancette, 2023-12-26:
We don't regularly test RViz on WSL2, so I don't think we have a lot of good advice here.  If you figure it out, please feel free to update this issue or add some documentation to http://docs.ros.org/en/rolling/How-To-Guides/Installation-Troubleshooting.html

OneCraftZRX, 2024-02-19:
just run `chmod 0700  /run/user/1000/` to fix it

surfaceowl, 2024-04-22:
the fix `chmod 0700  /run/user/1000/` is only temporary for that shell session, as the `1000` directory is recreated at reboot with 0755 permissions every time.  Perhaps the app should be changed to look for permissions of at least `0700`, rather than exactly `0700`, so that users with default permissions at `0755` for `1000` dir would pass.

RAYaRayaRay, 2024-05-21:
I'm having the exact same issue and I've searched everywhere, waiting for an answer.

nicholascabrera, 2024-05-27:
I added `chmod 0700 /run/user/1000/` to `.bashrc` and I haven't encountered the problem again

glangsto, 2025-01-11:
Adding chmod 0700 /run/user/1000 to .bashrc worked for me too.
Thanks!

CULTofBEETROOT, 2025-04-07:
> just run `chmod 0700 /run/user/1000/` to fix it

Hi... or perhaps: 

chmod 0700  /run/user/1000/

because this is about ownership and not rights?.

QuasarHeart, 2025-09-04:
According to the XDG spec, $XDG_RUNTIME_DIR (usually /run/user/$UID) must be 0700.
In some environments (e.g. WSL2), this directory is created as 0755.
Qt performs a strict permission check and logs this warning.
On WSL2 this is usually harmless, but it’s noisy.
Automatically fix the permissions on shell startup.
Add this line to your ~/.bashrc or ~/.zshrc:

chmod 0700 /run/user/$(id -u) 2>/dev/null
This ensures the directory has the expected permissions every time you start WSL.