ros2/rviz #1112: How to display topic frequence info on rviz2?

yang19900808, 2023-12-26:
Hello everyone, is it possible to display topic hz info on rviz2 just like command "ros2 topic hz xxx"?
If it is possible, how?
[image]

Yadunund, 2023-12-26:
I don't think we keep track of the frequency for a topic but this could be easily implemented by extending the definition of [RosTopicDisplay::incomingMessage](https://github.com/ros2/rviz/blob/rolling/rviz_common/include/rviz_common/ros_topic_display.hpp#L256) and managing a `subscription_start_time` member everytime `onEnable()/onDsiable()` is invoked.  You can then calculate the frequency by dividing the `messages_received_` by the elapsed time since subscription was started.

yang19900808, 2023-12-26:
> I don't think we keep track of the frequency for a topic but this could be easily implemented by extending the definition of [...]

It means that the only way to resolve my problem is to change the source code and build my own version of rviz2?

ahcorde, 2023-12-26:
> It means that the only way to resolve my problem is to change the source code and build my own version of rviz2?

Yes, this feature is not implemented, but you can consider contribute back this and open a PR. Then you don't need to compile your own version because this feautre will be available for all the community.

yang19900808, 2023-12-26:
> Yes, this feature is not implemented, but you can consider contribute back this and open a PR. Then you don't need to compile your own version because this feautre will be available for all the [...]

OK, thank you guys for answering my question.

ahcorde, 2023-12-26:
I'm closing this issue, but feel free to reopen it if you are going to work on this feature

Yadunund, 2023-12-27:
@yang19900808 I opened https://github.com/ros2/rviz/pull/1113 to add this feature.