ros2/rviz #1114: rviz2 via composition

mhubii, 2023-12-29:
From the CMake it appears the ros2_control_node is compiled into an executable, refer

https://github.com/ros2/rviz/blob/68d9167758b651d2e4c01f4aeb74a9613110e751/rviz2/CMakeLists.txt#L44

Are there plans to provide composition in the future?

roncapat, 2023-12-29:
Subscribing here. As soon as I get some free time I may give a look to evaluate the required effort to do so, but I would also like to know from the maintainers their opinion on such a possible development.

clalancette, 2023-12-29:
I'm not opposed to it, but I'd like to hear what the use case is.  I mean, yes, it will be more efficient to compose it with the thing it is debugging.  But the "typical" deployment is to have rviz2 on a separate machine from the robot, so I'd want to hear about the cases where that is not happening.

mhubii, 2023-12-30:
was just thinking that this could enable intra-process communication to rviz, although inter-process works just fine.

The `rclcpp_components_register_node`  cmake macro https://github.com/ros2/rclcpp/blob/5d2b32b5c3e5bbf538807779a7151ac1363002d2/rclcpp_components/cmake/rclcpp_components_register_node.cmake#L25C4-L25C4 lets users compile into a component and (I think) also compiles an executable.

roncapat, 2023-12-30:
@mhubii I think @clalancette is asking for a real use cases. I can propose a few to start with:

- simulations: for example, I have an E2E perception stack simulator that runs 100% composed; to view the point clouds in real time I might like to include Rviz in the component container & enable intra-process to avoid to slow down my simulation (because when I subscribe to a point cloud topic via classic Rviz, serialization/deserialization kicks in and I "loose" benefits of intra-process on that topic.
- post-processing of real robot data: suppose you receive some kind of data from a robot (which is a different machine, so no way to leverage composition here from the start) and you want to post-process it before display. This means you may need some control-station-side nodes for data processing, and you might like to compose them with Rviz
- playback: recently, I helped implement Rosbag2 composition. So, assume you have a large bag to play... Well, you might like to compose the rosbag2_transport::Player with Rviz to reduce the overhead displaying what is inside the bag.

Of course, I think these are not beginner setups, but may come handy.

roncapat, 2024-01-05:
Here is a little assessment of what's currently going on, and how the above objective coud be achieved.

As I was expecting, here we have a Qt Application along with its own startup logic.
The actual `rclcpp::Node` is created as an object by the application.

https://github.com/ros2/rviz/blob/68d9167758b651d2e4c01f4aeb74a9613110e751/rviz_common/include/rviz_common/ros_integration/ros_node_abstraction.hpp#L55-L64

https://github.com/ros2/rviz/blob/68d9167758b651d2e4c01f4aeb74a9613110e751/rviz_common/src/rviz_common/ros_integration/ros_node_abstraction.cpp#L46-L48

This is almost the same pattern used for example by `gazebo_ros_pkgs` and makes just sense.
It would be too complicated to transform Rviz in an `rclcpp_component` with a Qt GUI. 
What is can become, instead, is a Qt GUI with a `component_container`.

Let me exaplain this. Rviz could:
- host a `rclcpp_components::ComponentManager`
- attach its current node instance to the same executor of the component manager

Then, users could call the component manager services to do run-time composition (i.e. inject components in the Rviz instance).

I did the same locally with `gazebo_ros_pkgs` some time ago to circumvent `point_cloud` transform limitation. Useless to say, it worked very well (much faster fps).

ahcorde, 2024-01-25:
Hi @roncapat, 

Can you provide any pointer to your changes in gazebo_ros_pkgs ?

roncapat, 2024-02-07:
Here you are @ahcorde:
https://github.com/Fixit-Davide/gazebo_ros_pkgs/commit/a51a193806c3f144cba66954612f078ba1e0958b
https://github.com/Fixit-Davide/gazebo_ros_pkgs/commit/b93a2f3a1738104b90388ef216a7b376066ccc5b
CC: @Fixit-davide

ahcorde, 2024-02-15:
I created a draft PR https://github.com/ros2/rviz/pull/1140