ros2/rviz #1116: InteractiveMarkers Crashing On Rolling

AustinMooreT, 2024-01-04:
I ran into an unhandled std::runtime_error when using a rolling build of rviz2 on Windows.
I don't think Windows has anything to do with it, but thought it would worthwhile to mention my platform.

It seems that recently the way frame_locked_ was computed in interactive_markers.cpp has changed.
See fe52932 for reference

The exception is generated by the following change in that commit (and similar changes).

```diff
@@ -97,9 +97,9 @@ void InteractiveMarker::processMessage(
     orientation.w = 1.0;
   }
 
-  reference_time_ = message.header.stamp;
+  reference_time_ = rclcpp::Time(message.header.stamp, RCL_ROS_TIME);
   reference_frame_ = message.header.frame_id;
-  frame_locked_ = (message.header.stamp == rclcpp::Time());
+  frame_locked_ = (reference_time_ == rclcpp::Time());
 
   requestPoseUpdate(position, orientation);
   context_->queueRender();
```

I didn't follow up on what exactly frame_locked_ is used for, but the exception is thrown on the comparison 
```cpp
reference_time_ == rclcpp::Time()
```

The reason being reference_time_ is explicitly constructed with RCL_ROS_TIME for it's rcl_clock_type_t, but the default constructed rvalue rclcpp::Time() will have a rcl_clock_type_t being RCL_SYSTEM_TIME. This difference in clock types will cause an exception in rclcpp::Time::operator==.

Constructor for rclcpp::Time :
```cpp
RCLCPP_PUBLIC
explicit Time(int64_t nanoseconds = 0, rcl_clock_type_t clock_type = RCL_SYSTEM_TIME);
```
Comparison operator for rclcpp::Time :
```cpp
bool Time::operator==(const rclcpp::Time & rhs) const
{
  if (rcl_time_.clock_type != rhs.rcl_time_.clock_type) {
    throw std::runtime_error("can't compare times with different time sources");
  }

  return rcl_time_.nanoseconds == rhs.rcl_time_.nanoseconds;
}

```
See both
https://github.com/ros2/rclcpp/blob/rolling/rclcpp/include/rclcpp/time.hpp
and
https://github.com/ros2/rclcpp/blob/rolling/rclcpp/src/rclcpp/time.cpp
for reference.

This PR documents why the change was made
https://github.com/ros2/rclcpp/pull/2293
And this comment on the PR mentions not knowing if RCL_ROS_TIME or RCL_SYSTEM_TIME is the correct choice.
https://github.com/ros2/rclcpp/pull/2293#issuecomment-1800852519

To mitigate issues for myself I switched the explicit clock type from RCL_ROS_TIME to RCL_SYSTEM_TIME on my local, but am unsure if this is the appropriate fix.

ahcorde, 2024-01-05:
Hi @AustinMooreT, 

Thank you for explaining the issue, do you mind to open a PR ? We can discuss there which one is the best option.

AiVerisimilitude, 2024-01-08:
Hi,

This was an oversight on my end, thanks for taking a look at it! For fixing it, I'm not sure changing all the clocks to system time is the correct way to do it. Since the issue is with the comparison to default time of a message time stamp I would use the comparison operator of the builtin_interfaces::msg instead of rclcpp::Time eg.:

```cpp
frame_locked_ = (message.header.stamp == builtin_interfaces::msg::Time());
```

This should've been my default way of handling comparisons of message headers stamps with the empty timestamp.

Let me know if this sounds good. I will also see if I can find other instances of this typo.

AiVerisimilitude, 2024-01-11:
Should this issue be marked as closed, or how does the process go after merge of a fix?

clalancette, 2024-01-11:
Yeah, it should be closed.  Thanks for the report and the fix.

closed 2024-01-11 by clalancette