ros2/rviz #1122: [WARN]  New subscription discovered on topic '/scan', requesting incompatible QoS

ipa-hsd, 2024-01-12:
Setup: Ubuntu 22.04, ROS humble

I am writing a ROS driver for a laser scanner and want to view the scan data in RViz. I have used `SensorDataQoS` while creating the publisher
```cpp
scan_publisher_ = node_->create_publisher<sensor_msgs::msg::LaserScan>(scan_topic, rclcpp::SensorDataQoS());
```
and have also set the QoS values in .rviz as well
```yaml
      Topic:
        Depth: 5
        Durability Policy: Volatile
        Filter size: 10
        History Policy: Keep Last
        Reliability Policy: Best Effort
        Value: /scan
```
when I run the driver, I get the following warning
```
[ros_main-1] [WARN] [1700130825.414718492] [ros_main]: New subscription discovered on topic '/scan', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
```
But I can still view the data in RViz. Also, following is verbose output for the scan topic info:
```
ros2 topic info /scan --verbose
Type: sensor_msgs/msg/LaserScan

Publisher count: 1

Node name: ros_main
Node namespace: /
Topic type: sensor_msgs/msg/LaserScan
Endpoint type: PUBLISHER
GID: 01.0f.25.3c.d6.25.3e.66.01.00.00.00.00.00.12.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: BEST_EFFORT
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 1

Node name: rviz
Node namespace: /
Topic type: sensor_msgs/msg/LaserScan
Endpoint type: SUBSCRIPTION
GID: 01.0f.25.3c.da.25.19.b4.01.00.00.00.00.00.22.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: BEST_EFFORT
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite
```
So the settings do actually match. Why am I getting the warning in that case? And this happens only on humble (galactic and foxy works). I can do a test with the latest rolling as well, just in case, to narrow down that the problem is in the humble version.

audrow, 2024-01-25:
Hi @ipa-hsd, could you provide the full code for a minimal reproducible example? Also, which DDS are you using?

Thanks for making the issue.

wltjr, 2024-11-21:
I am experiencing this as well using pointcloud_to_laserscan, anytime I start Rviz2, it seems to be messing up the topic and pointcloud_to_laserscan stops publishing to the /scan topic. I am using what should be the latest available, ROS 2 Jazzy, Ubuntu Noble, packaged versions.
```bash
[pointcloud_to_laserscan_node-6] [WARN] [1732215357.520418540] [pointcloud_to_laserscan]: New subscription discovered on topic '/scan', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
[pointcloud_to_laserscan_node-6] [INFO] [1732215357.523424432] [pointcloud_to_laserscan]: Got a subscriber to laserscan, starting pointcloud subscriber
[pointcloud_to_laserscan_node-6] [INFO] [1732215482.382114669] [pointcloud_to_laserscan]: No subscribers to laserscan, shutting down pointcloud subscriber
```

I definitely changed it in Rviz2 to be best effort, to match the rest, I have no idea where the reliability policy is coming from. Like the original report, when I look at the topic information under ros2, rviz2 is using best effort. It seems to be happening on initial connect or something else in Rviz2.  I can see about providing additional information but not sure I can provide a reproducible example. Thank you!

daniel-zellmann-lowpad, 2025-01-13:
I have a similar issue where the QOS configuration in the RVIZ config file is not being applied. 
rviz version: ros-humble-rviz2 -> 11.2.14-1

Reproduce:
- start clean rviz2
- In `tool properties` change the QOS settings of the 2D pose estimate: 
[image]
- Store config file
- Restart rviz with config file
- In the `tool properties` the configuration is loaded correctly
- But with `ros2 topic info` it still shows "VOLATILE":
```
$ ros2 topic info /initialpose --verbose
Type: geometry_msgs/msg/PoseWithCovarianceStamped

Publisher count: 1

Node name: rviz
Node namespace: /
Topic type: geometry_msgs/msg/PoseWithCovarianceStamped
Endpoint type: PUBLISHER
GID: 01.10.d9.f0.32.96.8a.d4.c5.16.23.9e.00.00.1a.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): KEEP_LAST (5)
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 0
 ```
- In my application I assume the topic is Transient_Local, but is appears to be not: which is the reason I get the warning about "requesting incompatible QoS. No messages will be sent to it."

homemoon-cpu, 2025-01-25:
> I am experiencing this as well using pointcloud_to_laserscan, anytime I start Rviz2, it seems to be messing up the topic and pointcloud_to_laserscan stops publishing to the /scan topic. I am using what should be the latest available, ROS 2 Jazzy, Ubuntu Noble, packaged versions.
> 
> [pointcloud_to_laserscan_node-6] [WARN] [1732215357.520418540] [pointcloud_to_laserscan]: New subscription discovered on topic '/scan', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
> [pointcloud_to_laserscan_node-6] [INFO] [1732215357.523424432] [pointcloud_to_laserscan]: Got a subscriber to laserscan, starting pointcloud subscriber
> [pointcloud_to_laserscan_node-6] [INFO] [1732215482.382114669] [pointcloud_to_laserscan]: No subscribers to laserscan, shutting down pointcloud subscriber
> I definitely changed it in Rviz2 to be best effort, to match the rest, I have no idea where the reliability policy is coming from. Like the original report, when I look at the topic information under ros2, rviz2 is using best effort. It seems to be happening on initial connect or something else in Rviz2. I can see about providing additional information but not sure I can provide a reproducible example. Thank you!

I meet this problem on humble version. Do you know how to solve it now? Thanks!
```
[pointcloud_to_laserscan_node-9] [INFO] [1737804257.056397624] [pointcloud_to_laserscan]: Got a subscriber to laserscan, starting pointcloud subscriber
[pointcloud_to_laserscan_node-9] [INFO] [1737804270.573737362] [pointcloud_to_laserscan]: No subscribers to laserscan, shutting down pointcloud subscriber
```

JslBotBuilder, 2025-03-21:
I am facing this exact same problem!
`Type: sensor_msgs/msg/LaserScan

Publisher count: 1

Node name: pointcloud_to_laserscan
Node namespace: /mb_0
Topic type: sensor_msgs/msg/LaserScan
Endpoint type: PUBLISHER
GID: 01.10.1e.74.e4.ba.72.f4.df.6b.c1.fe.00.00.15.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: BEST_EFFORT
  History (Depth): KEEP_LAST (5)
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 4

Node name: amcl
Node namespace: /mb_0
Topic type: sensor_msgs/msg/LaserScan
Endpoint type: SUBSCRIPTION
GID: 01.10.6a.27.db.14.d7.ca.28.c5.e0.ca.00.00.6b.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: BEST_EFFORT
  History (Depth): KEEP_LAST (5)
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Node name: local_costmap
Node namespace: /mb_0/local_costmap
Topic type: sensor_msgs/msg/LaserScan
Endpoint type: SUBSCRIPTION
GID: 01.10.6a.27.db.14.d7.ca.28.c5.e0.ca.00.02.6c.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: BEST_EFFORT
  History (Depth): KEEP_LAST (50)
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Node name: global_costmap
Node namespace: /mb_0/global_costmap
Topic type: sensor_msgs/msg/LaserScan
Endpoint type: SUBSCRIPTION
GID: 01.10.6a.27.db.14.d7.ca.28.c5.e0.ca.00.02.76.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: BEST_EFFORT
  History (Depth): KEEP_LAST (50)
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Node name: rviz
Node namespace: /mb_0
Topic type: sensor_msgs/msg/LaserScan
Endpoint type: SUBSCRIPTION
GID: 01.10.bf.36.94.b1.3c.a3.e5.1d.01.29.00.00.4d.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: BEST_EFFORT
  History (Depth): KEEP_LAST (5)
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite
`
The only difference is the depth that both costmaps are asking for is different. However, the robot I'm launching has worked just fine in the past! Now I'm getting this issue and it's causing rviz not to recieve any of the laser scan data for navigation.

JslBotBuilder, 2025-03-21:
> > I am facing this exact same problem!
> > ```
> > Type: sensor_msgs/msg/LaserScan
> > 
> > Publisher count: 1
> > 
> > Node name: pointcloud_to_laserscan
> > Node namespace: /mb_0
> > Topic type: sensor_msgs/msg/LaserScan
> > Endpoint type: PUBLISHER
> > GID: 01.10.1e.74.e4.ba.72.f4.df.6b.c1.fe.00.00.15.03.00.00.00.00.00.00.00.00
> > QoS profile:
> >   Reliability: BEST_EFFORT
> >   History (Depth): KEEP_LAST (5)
> >   Durability: VOLATILE
> >   Lifespan: Infinite
> >   Deadline: Infinite
> >   Liveliness: AUTOMATIC
> >   Liveliness lease duration: Infinite
> > 
> > Subscription count: 4
> > 
> > Node name: amcl
> > Node namespace: /mb_0
> > Topic type: sensor_msgs/msg/LaserScan
> > Endpoint type: SUBSCRIPTION
> > GID: 01.10.6a.27.db.14.d7.ca.28.c5.e0.ca.00.00.6b.04.00.00.00.00.00.00.00.00
> > QoS profile:
> >   Reliability: BEST_EFFORT
> >   History (Depth): KEEP_LAST (5)
> >   Durability: VOLATILE
> >   Lifespan: Infinite
> >   Deadline: Infinite
> >   Liveliness: AUTOMATIC
> >   Liveliness lease duration: Infinite
> > 
> > Node name: local_costmap
> > Node namespace: /mb_0/local_costmap
> > Topic type: sensor_msgs/msg/LaserScan
> > Endpoint type: SUBSCRIPTION
> > GID: 01.10.6a.27.db.14.d7.ca.28.c5.e0.ca.00.02.6c.04.00.00.00.00.00.00.00.00
> > QoS profile:
> >   Reliability: BEST_EFFORT
> >   History (Depth): KEEP_LAST (50)
> >   Durability: VOLATILE
> >   Lifespan: Infinite
> >   Deadline: Infinite
> >   Liveliness: AUTOMATIC
> >   Liveliness lease duration: Infinite
> > 
> > Node name: global_costmap
> > Node namespace: /mb_0/global_costmap
> > Topic type: sensor_msgs/msg/LaserScan
> > Endpoint type: SUBSCRIPTION
> > GID: 01.10.6a.27.db.14.d7.ca.28.c5.e0.ca.00.02.76.04.00.00.00.00.00.00.00.00
> > QoS profile:
> >   Reliability: BEST_EFFORT
> >   History (Depth): KEEP_LAST (50)
> >   Durability: VOLATILE
> >   Lifespan: Infinite
> >   Deadline: Infinite
> >   Liveliness: AUTOMATIC
> >   Liveliness lease duration: Infinite
> > 
> > Node name: rviz
> > Node namespace: /mb_0
> > Topic type: sensor_msgs/msg/LaserScan
> > Endpoint type: SUBSCRIPTION
> > GID: 01.10.bf.36.94.b1.3c.a3.e5.1d.01.29.00.00.4d.04.00.00.00.00.00.00.00.00
> > QoS profile:
> >   Reliability: BEST_EFFORT
> >   History (Depth): KEEP_LAST (5)
> >   Durability: VOLATILE
> >   Lifespan: Infinite
> >   Deadline: Infinite
> >   Liveliness: AUTOMATIC
> >   Liveliness lease duration: Infinite
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > The only difference is the depth that both costmaps are asking for is different. However, the robot I'm launching has worked just fine in the past! Now I'm getting this issue and it's causing rviz not to recieve any of the laser scan data for navigation.
> 
> You can change the reliability of topic /scan to Best Effort in the rviz, then you can receive laser scan data normally.
> 
> [image]

It's already best effort in rviz. It still isn't showing the points.

JslBotBuilder, 2025-03-21:
[image]
Here is my topic in rviz

homemoon-cpu, 2025-03-21:
> [image] Here is my topic in rviz

Have you ever check whether your scan topic have data output

JslBotBuilder, 2025-03-21:
Ok scratch my case. I had an issue with my simulator not the package