ros2/rviz #1194: rviz2 MarkerArray transform problem

johannesn, 2024-05-03:
In our project we have a vehicle driving around, doing localization and regognizing objects in its surrounding as it goes. The localization determines how the vehicle moves through the world frame and thus publishes the transform from vehicle to world frame. The objects are published as a MarkerArray in the vehicle frame. The algorithm for recognizing objects takes less time than the localization, thus the MarkerArray is published before the transform from vehicle frame to odometry frame is ready. In rviz2 we use the world frame (i.e. odometry frame) as Fixed frame. Now rviz2 does not display the MarkerArray and reports a lot of errors of the kind "transform would require extrapolation into the future". However if we use multiple Marker's instead of the MarkerArray, the transform is successful. The transform is successfully applied to a multitude of messages (e.g. PointCloud2, PoseArray) however MarkerArray is the only message with problems with the transformation. 

In our project, we have a vehicle moving around, performing localization and object recognition in its surroundings as it progresses. Localization determines how the vehicle moves through the world frame and thus publishes the transform from the vehicle to the world frame. Objects are published as a MarkerArray in the vehicle frame. The object recognition algorithm takes less time than localization, so the MarkerArray is published before the transform from the vehicle frame to the odometry frame is ready. In rviz2, we use the world frame (i.e., odometry frame) as the Fixed frame. Now rviz2 does not display the MarkerArray and reports many errors of the kind "transform would require extrapolation into the future." However, if we use multiple Markers instead of the MarkerArray, the transform is successful. The transform is successfully applied to a multitude of messages (e.g., PointCloud2, PoseArray); however, MarkerArray is the only message with problems with the transformation.

Maybe this issue is related to another issue here: 
https://github.com/ros2/rviz/issues/363

I set up a sample project with sample data: 
https://github.com/johannesn/rviz_marker/blob/main/README.md

In the following pictures, the red boxes are Marker's, while the blue ones are MarkerArrays.
Successful transformation in same frame
[image]
Failing transformation in different frame
[image]

https://github.com/ros2/rviz/assets/6796392/0a2e6984-2022-404f-b29a-6c6c902c1a73

Gabry993, 2024-12-03:
Hello, are there any updates about this? I have the same issue with marker arrays published before tf is available. Is it possible that rviz in this case is not looking for the latest available transform, but it's using the timestamp from the array message?

johannesn, 2024-12-05:
Also still waiting for updates.

XinyuKhan, 2025-09-03:
MarkerArray msg does not have a header field, so it cannot be processed by a single marker. I'm going to implement this plugin by myself

peter-mitrano-ar, 2025-09-03:
@XinyuKhan if you do, consider making a Pull Request, or sharing you branch with your implementation here, so others can try it!

XinyuKhan, 2025-09-03:
@peter-mitrano-ar OK, let me have a try.