ros2/rviz #1198: Build Rviz for ROS2 fails, lots of ament_add_gmock errors

michaelbcd, 2024-05-13:
I build rviz of version 8.2.8 for arm ros2 in a docker with image ros:foxy --platform=arm64 (from https://discourse.ros.org/t/announcing-ros-foxy-docker-images-availability/14511), and after I fix some errors due to lack of third-party libraries, I am stucked in the following errors:
`
root@9e1882f604d9:/home/michael/work/src/rviz2# colcon build --merge-install
Starting >>> rviz_assimp_vendor
Starting >>> rviz_ogre_vendor
Finished <<< rviz_assimp_vendor [2.63s]
Finished <<< rviz_ogre_vendor [2.69s]
Starting >>> rviz_rendering
--- stderr: rviz_rendering
CMake Error at /opt/ros/foxy/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:69 (add_executable):
  Target "point_cloud_test_target" links to target
  "rviz_ogre_vendor::OgreOverlay" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:52 (_ament_add_gmock)
  CMakeLists.txt:186 (ament_add_gmock)

CMake Error at /opt/ros/foxy/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:69 (add_executable):
  Target "wrench_visual_test_target" links to target
  "rviz_ogre_vendor::OgreOverlay" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:52 (_ament_add_gmock)
  CMakeLists.txt:271 (ament_add_gmock)

CMake Error at /opt/ros/foxy/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:69 (add_executable):
  Target "string_helper_test" links to target "rviz_ogre_vendor::OgreOverlay"
  but the target was not found.  Perhaps a find_package() call is missing for
  an IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:52 (_ament_add_gmock)
  CMakeLists.txt:181 (ament_add_gmock)

CMake Error at /opt/ros/foxy/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:69 (add_executable):
  Target "point_cloud_renderable_test_target" links to target
  "rviz_ogre_vendor::OgreOverlay" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:52 (_ament_add_gmock)
  CMakeLists.txt:198 (ament_add_gmock)

CMake Error at CMakeLists.txt:171 (add_library):
  Target "rviz_rendering_test_utils" links to target
  "rviz_ogre_vendor::OgreOverlay" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?

CMake Error at /opt/ros/foxy/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:69 (add_executable):
  Target "billboard_line_test_target" links to target
  "rviz_ogre_vendor::OgreOverlay" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:52 (_ament_add_gmock)
  CMakeLists.txt:210 (ament_add_gmock)

CMake Error at CMakeLists.txt:67 (add_library):
  Target "rviz_rendering" links to target "rviz_ogre_vendor::OgreOverlay" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at /opt/ros/foxy/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:69 (add_executable):
  Target "covariance_visual_test_target" links to target
  "rviz_ogre_vendor::OgreOverlay" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
Call Stack (most recent call first):
[... 70 lines cut ...]
ros-foxy-ament-cmake-ros/focal,now 0.9.2-1focal.20230527.045739 arm64 [installed,automatic]
ros-foxy-ament-cmake-target-dependencies/focal,now 0.9.12-1focal.20230527.033914 arm64 [installed,automatic]
ros-foxy-ament-cmake-test/focal,now 0.9.12-1focal.20230527.033843 arm64 [installed,automatic]
ros-foxy-ament-cmake-uncrustify/focal,now 0.9.8-1focal.20230527.040848 arm64 [installed,automatic]
ros-foxy-ament-cmake-version/focal,now 0.9.12-1focal.20230527.033844 arm64 [installed,automatic]
ros-foxy-ament-cmake-xmllint/focal,now 0.9.8-1focal.20230527.041123 arm64 [installed,automatic]
ros-foxy-ament-cmake/focal,now 0.9.12-1focal.20230527.034003 arm64 [installed,automatic]
ros-foxy-ament-copyright/focal,now 0.9.8-1focal.20230527.040308 arm64 [installed,automatic]
ros-foxy-ament-cppcheck/focal,now 0.9.8-1focal.20230527.033907 arm64 [installed,automatic]
ros-foxy-ament-cpplint/focal,now 0.9.8-1focal.20230527.044409 arm64 [installed,automatic]
ros-foxy-ament-flake8/focal,now 0.9.8-1focal.20230527.034018 arm64 [installed,automatic]
ros-foxy-ament-index-cpp/focal,now 1.1.0-1focal.20230527.044824 arm64 [installed,automatic]
ros-foxy-ament-index-python/focal,now 1.1.0-1focal.20230527.040511 arm64 [installed,automatic]
ros-foxy-ament-lint-auto/focal,now 0.9.8-1focal.20230527.034012 arm64 [installed,automatic]
ros-foxy-ament-lint-cmake/focal,now 0.9.8-1focal.20230527.040514 arm64 [installed,automatic]
ros-foxy-ament-lint-common/focal,now 0.9.8-1focal.20230527.044618 arm64 [installed,automatic]
ros-foxy-ament-lint/focal,now 0.9.8-1focal.20230527.033909 arm64 [installed,automatic]
ros-foxy-ament-package/focal,now 0.9.5-1focal.20210422.231141 arm64 [installed,automatic]
ros-foxy-ament-pep257/focal,now 0.9.8-1focal.20230527.040159 arm64 [installed,automatic]
ros-foxy-ament-uncrustify/focal,now 0.9.8-1focal.20230527.034242 arm64 [installed,automatic]
ros-foxy-ament-xmllint/focal,now 0.9.8-1focal.20230527.040536 arm64 [installed,automatic]
ros-foxy-launch-testing-ament-cmake/focal,now 0.10.10-1focal.20230527.045532 arm64 [installed,automatic]
`
How to fix them?
Thanks.

ahcorde, 2024-05-13:
Hi @michaelbcd, 

foxy is [End of life](https://docs.ros.org/en/foxy/Releases.html#list-of-distributions), I would not expect that this should work. You should use an updated version of ROS 2.

In the future, please use one of the supported channels for help requests like this, such as the [Robotics Stack Exchange](https://robotics.stackexchange.com/). GitHub tickets should be used for bugs and enhancement tracking, not technical assistance.

closed 2024-05-13 by ahcorde

michaelbcd, 2024-05-13:
Hi @ahcorde,
OK, thanks. I searched this issue, and found some similar ones on this web, so I expected to get some answers or recommendations here.
I'll ask it on Robotics Stack Exchange [https://robotics.stackexchange.com](url).