ros2/rviz #1208: Laser Scan rotates with robot ------- [ERROR] [1716697697.080277840] [rviz2]: Lookup would require extrapolation into the future. Requested time 1716697697.040514 but the latest data is at time 1716697697.039929, when looking up transform from frame [laser_frame] to frame [odom] #4369

jainvrushabh5, 2024-05-26:
Required Info:

Operating System: Ubuntu 22.04 LTS
ROS2 Version: ROS2 Humble
The LaserScan rotating with robot when I use teleop_twist_keyboard to manually rotate the robot. I am using the rplidar_ros package to run the lidar I am unable to perform navigation and mapping I tried mapping

[Screencast from 05-14-2024 01:27:45 PM.webm](https://github.com/ros-navigation/navigation2/assets/118986898/5f2a272e-8178-4362-898f-9c12189c02c0)

Map when lidar frame is rotating with robot

Tf tree

[frames_2024-05-26_00.17.45.pdf](https://github.com/ros-navigation/navigation2/files/15446145/frames_2024-05-26_00.17.45.pdf)

audrow, 2024-06-06:
It's not clear to me that this is a bug in Rviz.

I think it'd be better to ask this question on the [Robotics Stack Exchange](https://robotics.stackexchange.com/).

closed 2024-06-06 by audrow

libpet-programming, 2025-03-12:
> Required Info:
>
> Operating System: Ubuntu 22.04 LTS ROS2 Version: ROS2 Humble The LaserScan rotating with robot when I use teleop_twist_keyboard to manually rotate the robot. I am using the [...]

Hi, I have a similar problem. Do you know how to solve this? Thank you very much.

A-C-Simon, 2026-08-18:
How was this solved