ros2/rviz #1220: [Bug] Material not showing after upgrade from Iron to Jazzy

firesurfer, 2024-06-24:
## Description

We currently try to upgrade our stack from iron to jazzy. We mostly use stl files in our model. With rviz in iron all meshes + materials are shown correctly. In jazzy all meshes based on a stl file are shown in the same light gray (see screenshot)

[image]

Additionally rviz produces (Only in case the model is loaded via the moveit planning scene) and endless stream of such errors:

> [rviz2-17] [ERROR] [1719212675.170806883] [rviz2]: Can't assign material Shape86Material to the ManualObject MeshShape_ManualObject7 because this Material does not exist in group General. Have you forgotten to define it in a .material script?
[rviz2-17] [ERROR] [1719212675.171172805] [rviz2]: Can't assign material Shape88Material to the ManualObject MeshShape_ManualObject8 because this Material does not exist in group General. Have you forgotten to define it in a .material script?
[rviz2-17] [ERROR] [1719212675.171424283] [rviz2]: Can't assign material Shape90Material to the ManualObject MeshShape_ManualObject9 because this Material does not exist in group General. Have you forgotten to define it in a .material script?
[rviz2-17] [ERROR] [1719212675.171593158] [rviz2]: Can't assign material Shape92Material to the ManualObject MeshShape_ManualObject10 because this Material does not exist in group General. Have you forgotten to define it in a .material script?

It is actually this stream of endless errors that makes it unusable. 

## System

- jazzy binary install
- rviz version: 14.1.1

## Additional information

For most of these meshes a material is set in the `<visual>` block such as `<material name="steel_blue"/>`

chama1176, 2024-07-11:
I am getting the same error in my environment.
The material is applied to cylinders, etc., but not to stl.

manadaniel, 2024-07-25:
As previously mentioned in the thread of issue #1236 I keep getting endless rviz2 logging errors even after the latest update to version 14.1.3 of rviz2 which included a fix for issue #1236, but not for me. In my case I the objects producing the continuous rviz2 log errors are collision objects that were added using moveit2's applyCollisionObject methods and they include .stl meshes (note: the objects are not defined in a URDF file). 

The typical error message is the following, with the numbers between "Shape" and "Material" as well as after "Object" being kind of an computed counter increasing over time:

`[rviz2-3] [ERROR] [1721749218.192680538] [rviz2]: Can't assign material Shape18906Material to the ManualObject MeshShape_ManualObject17738 because this Material does not exist in group General. Have you forgotten to define it in a .material script?`

I am guessing here, but it seems that it kind of _generates_ materials (like "Shape18906Material" from log output cited above) counting upwards the whole time and it looks for this material in some group "General" which I am not familiar with. 

My current setup is

- Ubuntu 24.04
- ROS2 Jazzy Debian Packages installed via apt
- Currently installed rviz2 version: `ros-jazzy-rviz2/noble,now 14.1.3-1noble.20240719.202006 amd64 [installed,automatic]`

Any ideas?

mrjogo, 2024-09-25:
I also got the same errors with rviz2 installed from apt (14.1.5), and I subsequently tried building 14.0.0 and 14.2.0 from source (as an overlay, without uninstalling the apt package), and got the same errors.

I'm only using the MotionPlanning and Motion Planning Tasks plugins, so not sure if it's somehow related to those.

DaniGarciaLopez, 2024-09-26:
Same here! Ubuntu 24.04 with Jazzy from Debian packages and rviz2 14.1.5 from binaries

Markus-Simonsen, 2024-11-05:
I had the same problem and found a solution by looking a bit around. And taking inspiration from this similar issue in Moveit [here](https://github.com/moveit/moveit2/issues/1164).

I fixed the error by adding one parameter to a function in as seen in this commit that i changed in my forked version :)
[here](https://github.com/ros2/rviz/compare/jazzy...Markus-Simonsen:rviz:jazzy)

mrjogo, 2024-11-05:
Nice catch! Could you make a pull request?

Markus-Simonsen, 2024-11-05:
Sure I will try, I have never tried making a pull request before though XD

PR #1503 (2025-06-27): Assign the geometry to the resource group "rviz_rendering" (backport #1502)
## Description

Fixes #1220 

### Is this user-facing behavior change?

### Did you use Generative AI?

### Additional Information

Because the author of PR #1292 doesn't react, i filed a new PR.
I hope that's alright.
<hr>This is an automatic backport of pull request #1502 done by [Mergify](https://mergify.com).
PR #1504 (2025-06-27): Assign the geometry to the resource group "rviz_rendering" (backport #1502)
## Description

Fixes #1220 

### Is this user-facing behavior change?

### Did you use Generative AI?

### Additional Information

Because the author of PR #1292 doesn't react, i filed a new PR.
I hope that's alright.
<hr>This is an automatic backport of pull request #1502 done by [Mergify](https://mergify.com).