ros2/rviz #1221: RVIZ2 not using GPU

ryan-roche, 2024-06-25:
I'm running RVIZ2 inside a docker container with GPU enabled. The container is running on the docker engine installed inside WSL2 (I manually installed the docker engine inside the WSL2 VM as I was not able to get my GPU working with Docker Desktop on Windows).

`nvidia-smi` and `glxgears` work inside the container, so the GPU driver is properly "exposing" itself to the container. However, RVIZ runs slowly and `nvidia-smi` does not show it in the process list. The only process reported by `nvidia-smi` as using the GPU (even when RVIZ is running) is XWayland. SOS!!!

payl-ampa, 2024-07-08:
Running Linux docker containers on Windows without WSL2 is not supported, but you appear to have solved that issue. 👍🏾 

You can force an application to run using the nvidia gpu by using [Nvidia's PRIME](https://wiki.archlinux.org/title/PRIME) on linux. Is there a specific reason you want to run rviz in a container? If you need reproducibility it may be simpler to write a setup script.

Takayuki9999, 2024-09-08:
[This question](https://robotics.stackexchange.com/questions/111677/gazebo-not-use-nvidia-gpu-when-i-installed-ros2-jazzy-by-using-docker-desktop-on) on StackExchange is helpful.
I had the same problem, but I solved it.
I followed the answer in that question and added additional options to docker and then Rviz2 started using the GPU.